esp-rs/espflash #18: Compatibility with Slower Board

anlumo, 2021-04-01:
Hi,

I have [this older board](https://randomnerdtutorials.com/esp32-built-in-oled-ssd1306/) and it fails to work with espflash, but works fine with esptool.py.

After a bit of debugging, I found the issue to be here:

https://github.com/esp-rs/espflash/blob/45d5034d376f9d267bbab7dea29250cb706194a3/espflash/src/connection.rs#L52

When I change the first sleep from 100ms to 300ms, everything works fine. I suspect that something on that board is slower than usual, since the original version worked fine on an M5Stack.

Changing this delay time shouldn't have any impact on other boards, just 200ms slower writes.

MabezDev, 2021-05-12:
Hi @anlumo , apologies for the delay in getting to this!

Do you have the rev0 esp32 by any chance? I think you should be able to find out with `esptool.py chip_id`.

If so I think we will have to add extra delays like you suggest, see https://github.com/espressif/esptool/blob/e5f94fa7ea60403e804d782115f440385e8a5edd/esptool.py#L556-L567

anlumo, 2021-05-12:
Doesn't look like that:

```
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

jessebraham, 2021-05-12:
```
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
```

I've got you covered! I'll take a look at this either tonight or tomorrow.

MabezDev, 2021-05-12:
Interesting, I also have the rev1:

```
esptool.py v3.0
Serial port /dev/ttyUSB1
Connecting........_____.....____
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

and can flash fine. Perhaps there is another issue with your setup?

Regardless, it would be good to add the work around for the rev0 boards.

anlumo, 2021-05-12:
It might be the flash chip itself, although it's a ESP-WROOM-32 module, which I'd guess is pretty standardized.

Note that I had to increase the delay to 400ms now to make it more reliable. With this value it works every time.

MabezDev, 2021-05-13:
Does flashing with esptool work? If that doesn't work either I suspect it could be your board.

anlumo, 2021-05-13:
It does work reliably with esptool.py.

jessebraham, 2021-09-22:
Is anybody who was previously having issues able to check if the problems persist with the most recent version of espflash?

PR #92 (2021-11-22): Retry failed connection attempts
~~This is built on top of #75, so I will convert it from a draft once that gets merged.~~

I believe this should resolve #18 and #90.

This PR emulates the connection logic performed by `esptool.py` and should help with connection issues on older devices. Essentially this adds two things:

1) Retry the connection until we connect to the device, or reach the attempt limit
2) Alternate between the default delay and a longer delay, which helps with older devices (eg. ESP32 rev 0)
closed 2021-11-22 by icewind1991