esp-rs/espflash #26: Flashing custom partition table & bootloader

ivmarkov, 2021-08-18:
The crate has a very nice support for describing custom partition tables. It is just that this functionality is not exposed yet to the outside world, and currently, a partition table with a single factory app is [always flashed](https://github.com/esp-rs/espflash/blob/2c86f48ea3a61624494f35add1ac590d4ca12138/espflash/src/chip/esp32.rs#L65).

We need a command-line syntax in the form of an option to read a partition table description from a file, so that one can flash their own partition tables. Ideally, the partition table file format should follow the one used in the [ESP-IDF CSV format](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html), so that we do not bring more confusion.

Regarding flashing a custom bootloader - this is less of a priority, but same here - we just need a command line option, where the user can provide their own elf or bin image of the bootloader (not sure which one would be more appropriate - perhaps elf, for symmetry with the app image).

Finally, having these two options (a custom bootloader and a custom partition table) brings the question: should it be possible to separately flash each of these (bootloader, partition table, app image), or shall we stick with the "platformio" approach, where all of these are flashed always together. Not sure what is the answer here.

jessebraham, 2021-09-07:
I have begun working on the custom partition table portion of this issue. I have an initial implementation ready and just need to do some testing/verification on hardware, then write some unit tests. Hopefully won't be long, though. Once this is complete I will implement the bootloader feature.

At present I am keeping the flashing process as-is just to minimize my pain while developing these features :grin: I do not feel strongly one way or another regarding the flashing process (it's adequate for my needs), so I would like to hear some other people's opinions.

@MabezDev @icewind1991 @georgik

MabezDev, 2021-09-08:
> Finally, having these two options (a custom bootloader and a custom partition table) brings the question: should it be possible to separately flash each of these (bootloader, partition table, app [...]

Perhaps in the future, we could leverage cargo fingerprinting to only flash the partition table and bootloader if they have changed. I think for now, flashing everything is the simplest way forward.

jessebraham, 2021-09-10:
With #40 and #41 merged this issue can likely be closed. I agree with @MabezDev that just doing everything in one go is the simplest option for now. If there is a need for splitting out this process it can always be done when it's needed.

closed 2021-09-12 by icewind1991