esp-rs/espflash #28: [Proposal] Remove Xargo/xbuild and only use build-std

MabezDev, 2021-08-18:
The rationale behind this is simple - the less we have to support the better.

* All tools *require* nightly, with `build-std` being the only path to stable (even if it might be far away)
* With the introduction of the `build-std-features` flag, AFAIK `build-std` now has feature parity with `Xargo` and `xbuild`
* Switching means we can rely on cargo even more, for example cargo can handle this bit of code and avoid us hard-coding target names: https://github.com/esp-rs/espflash/blob/4921f6e50e0c08dfa33e3fd371cfa4978215eb80/espflash/src/chip/mod.rs#L106-L113
(Note that that bit of code is mainly why I am opening this, as it broke when we started upstreaming std targets).

Thoughts @jessebraham @icewind1991 @ivmarkov ?

jessebraham, 2021-08-18:
I'm in favour of this personally.

`xargo` has been in maintenance mode since early 2018 due to the fact that its functionality was making its way into the official toolchain. The README in the `cargo-xbuild` repository even recommends using `build-std` instead.

ivmarkov, 2021-08-18:
> * All tools _require_ nightly, with `build-std` being the only path to stable (even if it might be far away)

I'm a bit confused here. As per our last meeting, in fact, the only reason why we require nightly (or *will* require nightly once 1.56 stable is out) is... `cargo build-std`. Everything else is/will be - in fact - stable? Or am I missing something?

That's what I meant during the meeting that `cargo build-std` is dragging us into nightly. With the other 2 esoteric features we use in cargo now stabilized, and with LibStd upstreamed, we don't need nightly that much anymore.

Link to the relevant topic [here](https://github.com/esp-rs/rust/discussions/67#discussioncomment-1185952) where it says "Targeting esp32-c3/h2 (RiscV) + ESP-IDF with Stock Stable Rust & Cargo".

Now, the above is not considering the xtensa patches, but I guess we can apply them on top of the stable compiler anyway?

ivmarkov, 2021-08-18:
Or did you mean by "tools" that *both* `cargo-xbuild` and `xargo` require nightly? Ooooh, if so, I see your point, and I agree!

MabezDev, 2021-08-18:
> Or did you mean by "tools" that both cargo-xbuild and xargo require nightly (the "tools")? Ooooh, if so, I see your point, and I agree!

Yes exactly that! Apologies, I should have worded that better.

PR #29 (2021-08-21): Remove references to Xargo and cargo xbuild
* Default to always use build-std.

Closes #28
closed 2021-08-21 by icewind1991