esp-rs/espflash #51: 1.0.0 fails to flash binary while esptool.py does

sehraf, 2021-09-21:
I tried flashing it a few times but it always fails:

<pre><font color="#4E9A06">└──&gt; </font><b>%</b> <font color="#4E9A06">~/.cargo/bin/espflash</font> <u style="text-decoration-style:single">/dev/ttyUSB0</u> <u style="text-decoration-style:single">target/riscv32imc-esp-espidf/debug/esphome-rs</u>
[00:00:01] <font color="#06989A">########################################</font>      12/12      segment 0x0
[00:00:00] <font color="#06989A">########################################</font>       1/1       segment 0x8000
Error: ────[<font color="#AA759F">espflash::timeout</font>]───────────────────────────────────────────────────────────────────────

    <font color="#FF0014">×</font> Communication error while flashing device
<font color="#FF0014">    ╰─▶ </font>Timeout while running command

</pre>

With esptool.py it works fine. 
Using `esptool.py --chip esp32c3 elf2image target/riscv32imc-esp-espidf/debug/esphome-rs && esptool.py --chip esp32c3 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x10000 target/riscv32imc-esp-espidf/debug/esphome-rs.bin`

It's an ESP32-C3
<pre><font color="#4E9A06">~/.cargo/bin/espflash</font> --board-info <u style="text-decoration-style:single">/dev/ttyUSB0</u> <u style="text-decoration-style:single">target/riscv32imc-esp-espidf/debug/esphome-rs</u>
Chip type: ESP32-C3
Flash size: 4MB
</pre>

I've previously used Arduino IDE to change the partition to "Huge App" (*3MB No OTA, 1MB SPIFFS*)
[binary.gz](https://github.com/esp-rs/espflash/files/7205390/binary.gz)

icewind1991, 2021-09-21:
https://github.com/esp-rs/espflash/pull/54 makes the flashing succeed for me but the device gives a bootloader error even when using what I assume is the [correct partition table](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/huge_app.csv)

sehraf, 2021-09-21:
Mhh my ESP prints this
>I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table

So the table differs ... also where is my SPIFFS? Not that i need it, just wondering.

---
I've reuploaded a sketch with Arduino IDE and now it is
>I (75) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (91) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (97) boot:  2 app0             OTA app          00 10 00010000 00300000
I (104) boot:  3 spiffs           Unknown data     01 82 00310000 000f0000
I (110) boot: End of partition table

Does that make any sense?

icewind1991, 2021-09-22:
https://github.com/esp-rs/espflash/pull/56 fixes this for me.

For the partition table you'll need to flash with `--partition-table huge_app.csv` [which you can find here](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/huge_app.csv)

icewind1991, 2021-09-23:
Should be fixed with 1.0.1

closed 2021-09-23 by icewind1991

sehraf, 2021-09-23:
can confirm, flashing works now!