esp-rs/espflash #52: Serial port auto-detection

igrr, 2021-09-21:
esptool.py implements serial port auto-detection logic, which lets users omit the port name in many cases. Similar logic could be implemented in this project.

Additionally, if the serial port is provided by a USB device and we can get enough information from the OS about the serial port to map it to a USB device, we can use VID:PID pairs to detect likely candidates for the development board serial ports. This should work pretty well e.g. for ESP32-C3 built-in USB_SERIAL_JTAG.

icewind1991, 2021-09-21:
The main reason I'm hesitant with doing this is that sending reboot+sync command to random serial devices might have unforeseen consequences.

*If* we can be reasonably sure that a device is an esp device than this might be fine

jessebraham, 2021-10-02:
I have begun working on this feature, should be pretty straight-forward.

One question I have is, how we should go about checking `VID`/`PID` values during the candidate serial port selection process?

I found the [espressif/usb-pids](https://github.com/espressif/usb-pids) repository which states that Espressif's `VID` is `0x303A` and lists a number of `PIDS`. I was thinking that for the auto-detection process we could only include these devices, which would somewhat address @icewind1991 's issue. Thoughts on this?

jessebraham, 2021-10-02:
How do we want to go about verifying VID/PID values for this?

I did find the [espressif/usb-pids](https://github.com/espressif/usb-pids) repository which lists Espressif's VID and a number of PIDs, however this does not include all boards (eg. a number use the Silicon Labs CP2102).

icewind1991, 2021-10-03:
I would generally err on the side of caution here, it's easy to tell the user to manually specify the port while trying to flash to a random device can cause larger issues.

My basic idea was:
- only try devices with from a known usb serial adapter
- only try to flash if there is a single matching device
  if there is more than one matching device, print out the options and tell the user to pick one manually

Adding a config option for adding additional VID/PID values would also be usefull

PR #75 (2021-11-19): Serial port auto-detection
Will eventually close #52.

Tested various chips and functionality on macOS, and minimal testing on Windows 10.

* The `serial` crate was swapped out with `serialport`
    * `serial` has not received an update in ~4 years, `serialport` is somewhat maintained
    * `serialport` was almost a drop-in replacement, and gives us the `available_ports` function as well
* Implements fairly basic port detection for `cargo espflash` and `espflash`
    * If no port has been specified by CLI or config, attempts auto-detection
    * If only one port is detected, it is selected aut [... cut]
closed 2021-11-19 by icewind1991