esp-rs/espflash #55: espflash 1.0 generates broken binary image (for ESP32 at least) and bootloops the board

ivmarkov, 2021-09-22:
* Build the [rust-esp32-std-hello]() crate as per the instructions
* Install `espflash 1.0` with `cargo install espflash`
* Flash with `espflash /dev/tty.usbserial-0204D0A0 target/xtensa-esp32-espidf/debug/rust-esp32-std-hello`
* Open up a monitor: `espmonitor /dev/tty.usbserial-0204D0A0`

Output:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
E (94) esp_image: Segment 2 load address 0x3f40fff8, doesn't match data 0x000334e8
E (94) boot: Factory app partition is not bootable
E (96) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
...
```

icewind1991, 2021-09-22:
Can you try if https://github.com/esp-rs/espflash/pull/56 fixes this for you

ivmarkov, 2021-09-22:
It does!

What can I say... party time! *<(:)

closed 2021-09-23 by icewind1991