esp-rs/espflash #62: The system cannot find the path specified. (os error 3)

ajsb85, 2021-09-30:
```console
.\cargo-espflash.exe espflash --board-info
Error: ─────────────────────────────────────────────────────────────────────────────────────────────────────────────────

    × Failed to parse Cargo.toml
    ├─▶ The system cannot find the path specified. (os error 3)
    ╰─▶ The system cannot find the path specified. (os error 3)
```

ivmarkov, 2021-09-30:
Your command line should just be: `espflash --board-info`

icewind1991, 2021-09-30:
Are you running the command from within the project? (next the `Cargo.toml`)

ajsb85, 2021-09-30:
Hi, @ivmarkov 

```console
PS C:\Users\ajsb85> espflash --board-info
Usage: espflash [--board-info] [--ram] [--partition-table partition.csv] [--bootloader boot.bin] <serial> <elf image>
```

Hi, @icewind1991 

```console
$ cat .\Cargo.toml
[package]
name = "lvgl_demo"
version = "1.0.0"

[connection]
serial = "COM16"
```

```console
$ cargo-espflash.exe espflash .\firmware.bin
Error: ────[cargo_espflash::build_std]──────────────────────────────────────────────────────────────────────────────────
    × 'build-std' not configured

    ‽ cargo currently requires the unstable 'build-std' feature, ensure that .cargo/config{.toml} has the
      appropriate options.
                See: https://doc.rust-lang.org/cargo/reference/unstable.html#build-std

```

ajsb85, 2021-09-30:
Holding boot push button before start.
And using the .elf
And it is works!

```console
espflash COM13 .\firmware.elf
[00:00:03] ########################################      16/16      segment 0x1000
[00:00:00] ########################################       1/1       segment 0x8000
[00:01:04] ########################################     280/280     segment 0x10000
```

closed 2021-09-30 by ajsb85