esp-rs/espflash #70: Support flash encryption

jessebraham, 2021-10-06:
Some chips support flash encryption (for example, the [ESP32-S2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/flash-encryption.html)). This should be an option when flashing.

jessebraham, 2022-07-06:
After some further discussion, it seems there's not really anything actionable here. I have written a short guide for using flash encryption in this gist:  
https://gist.github.com/jessebraham/0c7d86f9866614dc602e5d8e0e97c37a

If there is any specific functionality we need in the future we can open issues for that specifically.

closed 2022-07-06 by jessebraham

ostenning, 2024-05-23:
@jessebraham in your gist, you need to manually build a secure flash bootloader with esp-idf idf.py, and then manually pass that bootloader binary to espflash. 

That seems fairly incomplete to me. Shouldn't `espflash` be able to prepare the bootloader for encryption?