esp-rs/espflash #74: Not working in windows

stevefan1999-personal, 2021-10-18:
Hi, I wonder why it is not doing okay in Windows? I'm using [nanoESP32-C3](https://github.com/wuxx/nanoESP32-C3) which should be compatible, but:
```pwsh
PS E:\Git\github.com\MabezDev\esp32c3-experiments> cargo espflash --speed 115200 COM4
Serial port: COM4
Connecting...

Error: espflash::connection_failed
  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down

```
Which I can flash correctly with esptool

mladedav, 2021-12-05:
This exact error is printed when you're not in flashing mode, i.e. you didn't restart the device while G0 connected to GND. I don't know if that's your issue (since you're able to flash with esptool) but maybe this can help someone.

JurajSadel, 2022-02-08:
Hi @stevefan1999-personal, did [this]( https://github.com/esp-rs/espflash/commit/85c8eb84db7f0a534d73772069f864475ff4dfe6) resolve your issue or you are still having the same issue?

closed 2022-03-25 by jessebraham