esp-rs/espflash #78: "elf2image" subcommand in espflash

ivmarkov, 2021-10-20:
I keep trying to eradicate `esptool.py` from my demo crate build instructions, yet I'm constantly returning to it, because of small missing functionality in `espflash`. Case in point: conversion from ELF to BIN.

I do realize this conversion is available in `cargo-espflash` - yet - using `cargo-espflash` is not an option until it gets all build parameters that plain `cargo` has. For one, how do I pass `--features x,y` to `cargo-espflash`? Furthermore, I'm not really convinced that all works well with `cargo-espflash`. Trying to build my demo crate with it fails as if it can't see the `build-std` flags in `.cargo/config.toml`.

So how about folding or copying the ELF to BIN functionality in eslflash? I do realize this might require specifying the chip for which the binary will be produced, yet this is a small price to pay for the extra flexibility...?

jessebraham, 2021-10-20:
I'll come back to this issue when I have a moment and address the other points, but you should be able to use the `--features` flag directly with `cargo espflash`:

```shell
$ cargo espflash --features x,y save-image image.bin
```

If this is not working then there is indeed a bug somewhere :)

ivmarkov, 2021-10-20:
By the way, I don't even see the `save_image` option when issuing `cargo espflash --help`. And in general, the help screen is pretty spartan, so to say.

Have you guys considered putting in some `structopt` based argument parsing and command line documentation? We use it [here](https://github.com/ivmarkov/embuild/blob/master/cargo-pio/src/main.rs#L16) and it is simply beautiful.

jessebraham, 2021-10-20:
I think once #77 is complete (it is already in-progress) then it should be fairly trivial to have `save-image` available for both applications. If this is not already the case when that issue is closed I will PR the required changes.

I will begin working on the image merging feature this week.

closed 2021-11-02 by jessebraham