esp-rs/espflash #85: Cannot connect to serial device

imxood, 2021-10-30:
The serial device cannot be found using espflash. However, the serial port can be opened by using espminitor
[image]

Just bought the esp32c3, ready to burn the program for the first time

Taobao link: https://item.taobao.com/item.htm?spm=a1z09.2.0.0.7eea2e8dJZTw4u&id=652515479052&_u=u1fh48sn6cb5

The compiled project is based on https://github.com/ivmarkov/rust-esp32-std-mini.git, simple modification

pavlus, 2022-01-21:
May be related to #116

jessebraham, 2022-04-06:
I'm going to close this due to inactivity. If you're able to test again using the newest version (`1.4.0`) and this problem persists, please feel free to re-open the issue.

closed 2022-04-06 by jessebraham