esp-rs/espflash #93: builtin serial/jtag - unable to flash via serial

MabezDev, 2021-11-19:
The builtin serial/jtag peripheral of the c3 & s3 also exposes a serial port.

With recent versions of esptool it is possible to flash via the serial port, however it is not possible to flash with espflash currently (it hangs).

To reproduce:

1. Wire up a USB cable to the serial/jtag peripheral pins.
2. lsusb should report something like this:
    ```
    Bus 000 Device 010: ID 303a:1001 303a USB JTAG/serial debug unit  Serial: 7C:DF:A1:61:E4:80
    ``` 
3. A serial port should also show up, on mac/linux it comes up under `/dev/ttyX` (windows untested)
4. Try and use espflash over this port

jessebraham, 2021-11-20:
Referencing [esptool.py](https://github.com/espressif/esptool/blob/master/esptool.py) it seems there are a number of procedures which differ between the standard flashing process and using JTAG. As we do not presently support flashing over JTAG (see #23) this behaviour is mostly expected.

Thank you for bringing this (back) to attention though, we'll get to this eventually I'm sure :)

igrr, 2021-11-20:
I think esptool doesn't implement flashing over JTAG, it flashes over the "serial" part of the USB-SERIAL-JTAG peripheral. The difference between that and flashing over UART is in the reset / download mode entry procedure. It still uses DTR/RTS pins but the sequence in which they have to be set is slightly different compared to normal case.

The changes required are in this commit: https://github.com/espressif/esptool/commit/5f60add5ccf910a4ed51065b095f1de580831945

PR #122 (2022-01-27): Allow for flashing via the USB Serial JTAG peripheral
Thanks to @JurajSadel for his work on this!

As stated in the title, this adds support for flashing via the ESP32-C3's USB Serial JTAG peripheral. This required performing a small refactor for `get_serial_port` to instead return a `SerialPortInfo` struct, as we needed access to the `PID` later on.

Has been tested on macOS x86_64 successfully. I will test it on Windows later today, and I believe @MabezDev will be able to test on an M1 Mac.

Closes #93.
closed 2022-01-27 by jessebraham