esp-rs/espflash #95: How to set flash size

Verequies, 2021-11-21:
I'm currently trying to change the flash size from 2 MB to 8 MB as I am getting the following warning:
```
W (269) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
```
However I am unsure how to do this properly. Changing the flash size in the `sdkconfig` does not work either.

jessebraham, 2021-11-22:
I believe (and somebody please correct me if I'm wrong) this is because of the default bootloaders we've included in `espflash`. Flash size can be set prior to compile time via `menuconfig`, but since we do not build the bootloader when compiling the application there is presently no way to change this value.

As a workaround, you should be able to build the `bootloader.bin` file using existing methods (eg. `esp-idf` and `idf.py`), then tell `espflash` to use it instead by passing the `--bootloader bootloader.bin` argument.

jessebraham, 2022-03-11:
I was indeed mistaken, this value is being specified in the application binary header. You should be able to specify `--flash-size` on the command-line now that #141 has been merged, this will be included in the next release.

closed 2022-03-11 by jessebraham