esp-rs/espflash #101: Error is not helpful when argument is misplaced

mamins1376, 2021-11-29:
I was trying to find out why my firmware does not work, and tried to look what's in it, so after reading `cargo espflash --help`, I changed this:
```
$ cargo espflash --example ex /dev/ttyUSB0
```
to this:
```
$ cargo espflash --example ex save-image out.bin
Error: cargo_espflash::no_artifact

  × No executable artifact found
  help: If you're trying to run an example you need to specify it using the `--example` argument
        or if you're in a cargo workspace, specify the binary package with `--package`.
```

And after looking into the code and trying to debug that, found out I just misplaced the `--example ex` flag:
```
$ cargo espflash save-image --example ex out.bin
```

jessebraham, 2021-12-08:
Thank you for reporting this.

This is an unfortunate side effect of how we have structured our command-line interface. Omitting a subcommand will invoke the default build and flash action, and as such we need to accept various options at the root level (ie. `cargo espflash [OPTIONS]`). Unfortunately there is some overlap between the options required to build and flash, and the options required for the various subcommands, so it's not immediately clear where to provide them.

For the time being, try to be specific as possible. If you're using a subcommand then pass the options to the subcommand instead of to the application itself. I'm not sure at present what the best approach to fixing this is, but I will look into it.

jessebraham, 2022-01-27:
I'm closing this due to inactivity, please feel free to re-open if you are not satisfied with the above response.

closed 2022-01-27 by jessebraham