esp-rs/espflash #103: Error message is unclear when user does not have permissions to access /dev/tty*

rishubn, 2021-12-11:
When the user does not have permissions for `/dev/tty*`, the error indicates connection failure. While technically correct, the error should present a suggestion to check user permissions:

```
espflash /dev/ttyUSB0 target/riscv32imc-esp-espidf/debug/esp32-hellow
Error: espflash::connection_failed

  × Failed to open serial port /dev/ttyUSB0
  ├─▶ Error while connecting to device
  ╰─▶ Serial port not found
  help: Ensure that the device is connected and your host recognizes the serial
        adapter
```

The above error was fixed by adding `$USER` to the `dialout` group:

`sudo  useradd -aG dialout $USER` followed by reboot/logout

icewind1991, 2021-12-12:
When I remove permissions for the tty and try it I do get the expected `Permission denied` error.

can you provide some info about your host system?

jessebraham, 2022-01-27:
Closing this due to inactivity, please feel free to re-open if you feel there is actually a problem here.

closed 2022-01-27 by jessebraham