esp-rs/espflash #104: `partitions.csv` rejected with blank offset fields

bugadani, 2021-12-13:
According to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html it is perfectly valid to leave the offset fields blank. However, espflash rejects these `csv`s as malformed.

jessebraham, 2021-12-14:
Thanks for noticing this and opening a PR!

Just for context, I had initially left this implemented (see [this comment](https://github.com/esp-rs/espflash/blob/master/espflash/src/partition_table.rs#L407-L408)) out of laziness, but with the intention of coming back to it. As you can see, that never happened :)

bugadani, 2021-12-14:
As you no doubt has seen, I have even deleted that comment in my PR ;) But I'm more lazy to update (and maintain) my partition table, so I hope my solution is good enough to be merged :)

PR #106 (2021-12-15): Support multiple storage partitions and empty partition offsets
Closes #104 
Closes #105
closed 2021-12-15 by icewind1991