esp-rs/espflash #108: error: failed to compile `espflash v1.2.0`

Raunak-Singh-Inventor, 2021-12-16:
I am trying to run the [rust-esp32-std-demo](https://github.com/ivmarkov/rust-esp32-std-demo#flash). When I try to run the  `cargo install espflash`, the command errors out and fails to install espflash.

version: `v1.2.0`
operating system: `Linux x86_64`
cargo version: `cargo 1.57.0-dev`

Error:
```
error: failed to run custom build command for `libudev-sys v0.1.4`

Caused by:
  process didn't exit successfully: `/tmp/cargo-installR57Yug/release/build/libudev-sys-872bfa2eedb0d711/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "`\"pkg-config\" \"--libs\" \"--cflags\" \"libudev\"` did not exit successfully: exit status: 1\nerror: could not find system library 'libudev' required by the 'libudev-sys' crate\n\n--- stderr\nPackage libudev was not found in the pkg-config search path.\nPerhaps you should add the directory containing `libudev.pc'\nto the PKG_CONFIG_PATH environment variable\nNo package 'libudev' found\n"', /home/raunaksingh/.cargo/registry/src/github.com-1ecc6299db9ec823/libudev-sys-0.1.4/build.rs:38:41
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to compile `espflash v1.2.0`, intermediate artifacts can be found at `/tmp/cargo-installR57Yug`

Caused by:
  build failed
```

Raunak-Singh-Inventor, 2021-12-17:
fixed this error using this command: `sudo apt-get install libudev-dev`

closed 2021-12-17 by Raunak-Singh-Inventor

maspetsberger, 2022-02-27:
The equivalent package on Fedora 35 is `systemd-devel`, for other distributions check: https://pkgs.org/download/libudev-devel

jessebraham, 2022-02-28:
Thanks, I've added a [section](https://github.com/esp-rs/espflash#notes-on-building) to the README stating this requirement.

tomasApo, 2022-07-14:
@Raunak-Singh-Inventor thank you for the solution!

manupawickramasinghe, 2022-11-28:
Thank you for the solution