esp-rs/espflash #126: Flashing not working

luukvankooten, 2022-01-28:
```zsh
$ espflash /dev/cu.usbserial-0001 target/xtensa-esp32-espidf/release/led
Serial port: /dev/cu.usbserial-0001
Connecting...

Unable to connect, retrying with extra delay...
Unable to connect, retrying with default delay...
Unable to connect, retrying with extra delay...
Unable to connect, retrying with default delay...
Unable to connect, retrying with extra delay...
Unable to connect, retrying with default delay...

Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ReadReg command
```

Every single time I run espflash this output is sent to me. It looks like it's doing something because my previous program which is flashed stops. Looked into the issues, it is like #85 but not sure.

luukvankooten, 2022-01-30:
Got a workaround: install espflash with this specific comment: `cargo install espflash --git https://github.com/esp-rs/espflash.git --rev fd205c7d21cf7eb339ccb76d270d1821577c0254 --force`

pyaillet, 2022-02-06:
I faced the same problem. Reverting to v1.1.0 solved it for me.

I can provide

jessebraham, 2022-04-06:
Would either or both of you be able to check whether this issue persists using the newest version (`1.4.0`) please?

pyaillet, 2022-04-06:
I switched to 1.4.0-dev last week to test panic backtrace of `espmonitor` and did not face any problems since then.
I will try with the `1.4.0` release though.

pyaillet, 2022-04-07:
So I just installed the released `1.4.0` version and it works perfectly on my side.

Thank you for this update :)

closed 2022-04-11 by luukvankooten