esp-rs/espflash #129: Serial port not found on macOS

sirhcel, 2022-01-31:
- The changes introduced with 85c8eb8 break serial port detection on macOS 12 (Monterey)
- This currently gives the following output when attempting to access a ESP32-C3-DevKitM-1:
    ```
    espflash $ cargo run --bin espflash -- board-info /dev/tty.usbserial-113110
        Finished dev [unoptimized + debuginfo] target(s) in 0.04s
         Running `target/debug/espflash board-info /dev/tty.usbserial-113110`
    Error: cargo_espflash::no_serial

      × No serial port specified in arguments or config
      help: Add a command line option with the serial port to use
    ```

- While the device is present
    ```
    espflash $ ls -l /dev/tty.usbserial-113110                                                                        (master)
    crw-rw-rw-  1 root  wheel  0x900007a 31 Jan 23:47 /dev/tty.usbserial-113110
    ```

- It looks like this is caused by `serialport-rs` currently returning USB serial devices as `PciPort`
    - It returns actually available devices as `PciPort` instead of `UsbPort` and so the detection from [`detect_usb_serial_ports`](https://github.com/esp-rs/espflash/blob/040fe78054506c063e713ca427bf5b57e2290a33/espflash/src/cli/serial.rs#L129) returns an empty vector
    - USB serial devices are still returned from [`serialport::available_ports()`](https://github.com/esp-rs/espflash/blob/040fe78054506c063e713ca427bf5b57e2290a33/espflash/src/cli/serial.rs#L131) but as `PciPort`.

sirhcel, 2022-01-31:
- There are already two tickets tracking this issue at `serialport-rs`
    - https://gitlab.com/susurrus/serialport-rs/-/issues/106
    - https://gitlab.com/susurrus/serialport-rs/-/issues/115
- And there is the merge request https://gitlab.com/susurrus/serialport-rs/-/merge_requests/104 which fixes this issue for me

- Using this fixed version (https://github.com/sirhcel/espflash/commit/5d714448c7df25259625b3adcfec7968d49a66a0) of `serialport-rs` lets my development board show up again:
    ```
    espflash $ cargo run --bin espflash -- board-info /dev/tty.usbserial-113110
        Finished dev [unoptimized + debuginfo] target(s) in 0.04s
         Running `target/debug/espflash board-info /dev/tty.usbserial-113110`
    Serial port: /dev/tty.usbserial-113110
    Connecting...

    Chip type:         ESP32-C3 (revision 3)
    Crystal frequency: 40MHz
    Flash size:        4MB
    Features:          WiFi
    MAC address:       [...]
    ```

jessebraham, 2022-02-01:
Thanks for the issue.

I'll keep this brief, as I have made a longer comment in this issue:
https://gitlab.com/susurrus/serialport-rs/-/issues/111

We are working on getting the `serialport-rs` project back on its feet, whether that means taking over the project or forking it. I have been in discussion with some people regarding, so hopefully we will have a resolution in the coming weeks. We can then begin to address the upstream issues which are affecting us.

sirhcel, 2022-02-01:
Thank you for raising your hand at https://gitlab.com/susurrus/serialport-rs/-/issues/111! I was not aware of this issue when fiddling around with this issue here yesterday. This is definitely a critical path, so count me in as a maintainer deputy. Fingers crossed that things sort out well!

jessebraham, 2022-02-15:
No real updates regarding `serialport-rs` yet other than "I'm working on it". Hoping to have something usable by next week, but no promises.

I did apply a patch to my fork which seems to have solved this issue though! So once this fork is available we should have a solution for this issue.

closed 2022-04-06 by jessebraham