esp-rs/espflash #135: espflash 1.3 reports "No serial port specified"

mykmelez, 2022-02-18:
Perhaps this is a duplicate of #129, but that issue is about cargo-espflash, while this one is about espflash, so I thought I'd file it separately just in case.

With espflash version 1.2, I'm able to flash (and monitor) a binary as expected:

```sh
> cargo install --version 1.2.0 espflash
…
> espflash --version
espflash 1.2.0
> espflash /dev/cu.usbserial-2114401 target/xtensa-esp32s3-espidf/debug/example
Serial port: /dev/cu.usbserial-2114401
Connecting...
```

But version 1.3 complains that it can't find the serial port (despite being given the same command line args, including the serial port):

```sh
> cargo install espflash
…
> espflash --version
espflash 1.3.0
> espflash /dev/cu.usbserial-2114401 target/xtensa-esp32s3-espidf/debug/example
Error: cargo_espflash::no_serial

  × No serial port specified in arguments or config
  help: Add a command line option with the serial port to use
```

mykmelez, 2022-02-19:
Good guesses, I'm running macOS Monterey *and* using an M1!

Unfortunately, switching to that branch didn't seem to help:

```sh
> cargo install --git https://github.com/jessebraham/espflash/ --branch fixes/serialport espflash
…
Replaced package `espflash v1.3.0` with `espflash v1.4.0-dev (https://github.com/jessebraham/espflash/?branch=fixes/serialport#bea29ea5)` (executable `espflash`)
> espflash --version
espflash 1.4.0-dev
> espflash --monitor /dev/cu.usbserial-2114401 target/xtensa-esp32s3-espidf/debug/example
Error: cargo_espflash::no_serial
```

jessebraham, 2022-02-19:
That's unfortunate, thanks for checking at least. I will look into this ASAP!

stevecheckoway, 2022-02-19:
I just ran into this myself using macOS Monterey on an M1.

It seems like there are two problems
1. The error message is confusing because a serial port was passed on the command line but the error message says it wasn't.
2. `espflash` doesn't try to open user-specified path as a serial port.

For 2, it makes sense to me to only look at USB serial ports when looking for default serial ports (if that happens) but if a user passes a path to a serial port, it should be used as specified and if the serial port cannot be opened, then report the error at that point.

(My thinking is that this is similar to how one shouldn't check permissions to open a file via [access(2)](https://man7.org/linux/man-pages/man2/access.2.html) before opening file and instead should just try to open the file and check if it fails.)

mykmelez, 2022-02-19:
I just tried this on an Intel Mac running Monterey, and I experienced the same issue. Then @stevecheckoway's comment reminded me to try specifying `/[dev/tty.usbserial-2114401](http://dev/tty.usbserial-2114401)` instead of `/[dev/cu.usbserial-2114401](http://dev/cu.usbserial-2114401)`, and that worked:

```sh
> espflash --monitor /dev/tty.usbserial-14114401 target/xtensa-esp32s3-espidf/debug/example
Serial port: /dev/tty.usbserial-14114401
Connecting...
```

stevecheckoway, 2022-02-19:
That's surprising. Both the tty and the cu devices fail in exactly the same way for me.

jessebraham, 2022-02-22:
After a brief investigation today, it seems that `/dev/tty.*` devices are being listed by `serialport::available_ports()`, however the `/dev/cu.*` ports are not. This was done on an x86_64 mac running Big Sur, just for completeness.I don't believe this to be the full cause of the issue, but I was able to reproduce @mykmelez 's experience which led me to this. The short-term solution of course is just to use the `tty` ports, however this will need to be investigated in the `serialport` library; I'm not sure if there is a technical reason for this or not.

There seem to be some issues regarding both M1 and Monterey, so I will continue to look into this.

Firstyear, 2022-02-26:
M1 + Monterey, using `/dev/cu.usbserial-0001`  or  `/dev/tty.usbserial-0001` does not work.

Firstyear, 2022-02-26:
A bit of poking shows that detect_usb_serial_ports is returning an empty vector, so I'm wondering if something else in the PR is filtering the valid ports incorrectly. Sorry I can't help more.

Lainera, 2022-02-28:
Similar issue on Intel BigSur. 
Using `/dev/tty.usbserial-xyz` instead of `/dev/cu.usbserial-xyz` worked.

JensPfeifle, 2022-02-28:
I also experienced this issue running in a Podman container under Linux.

On the host: :heavy_check_mark: 
```shell
espflash /dev/ttyACM0 board-info
✔ Use serial port '/dev/ttyACM0' - Sunrise Point-LP USB 3.0 xHCI Controller? · yes
✔ Remember this serial port for future use? · yes
Serial port: /dev/ttyACM0
```

`sudo podman run --device /dev/ttyACM0 -v -it docker.io/espressif/idf-rust`

In the container: :x: 

```shell
espflash /dev/ttyACM0 board-info
Error: cargo_espflash::no_serial

  × No serial port specified in arguments or config
  help: Add a command line option with the serial port to use
```

jessebraham, 2022-03-03:
Still working on this, as alluded to in my previous comment there seem to be a number of factors at play here. I have reworked the error handling in the `cli::serial` module now and created a second error type to distinguish from `Error::NoSerial`, so this will hopefully make things more clear.

---

@Firstyear thank you for digging into this. There is a bug in the current version of `serialport` for Monterey which lists USB serial ports as PCI ports, so I suspect this is happening and they are consequently being filtered out of the results. We have established a [fork of serialport](https://github.com/serialport/serialport-rs) to which I have applied a fix for this problem, and I'm hoping to release a new version of this library soon.

---

@Lainera this is due to the `serialport` dependency (or possibly one of its dependencies), but I plan on seeing if this can be fixed. If not I will be sure to note it in the README at the very least, or try to come up with another solution.

---

@JensPfeifle not sure what is going on there but I'll try to reproduce it locally and dig in, thanks for reporting.

Firstyear, 2022-03-03:
Thanks for investigating this further, and for your time and work on this crate! Really appreciate it mate 😁

jessebraham, 2022-04-04:
Thank you everybody for being patient. I have released a new version of `serialport`, and have bumped the dependency in the `master` branch of this repo. I believe this release should resolve at least some of the issues we're experiencing.

I will need some more time to analyze what remaining problems need addressing, but hopefully we can get `1.4.0` of `espflash` out soon. If anybody is able/willing to test out the `master` branch and report back their results it would be very helpful!

jessebraham, 2022-04-06:
I have release `1.4.0` today, which I'm hoping will resolve at least some of the issues. If anybody is able to test using this version and report back on the status of the issues they were previously facing, I'd appreciate it!

jessebraham, 2022-04-13:
There have been quite a few comments in this issue spread over the course of a number of weeks, so I will try to summarize things the best that I can. I'm hoping that the problems reported in this issue have been resolved as of `1.4.0`, and would like to get this issue closed when possible.

It seems there were actually a number of issues:

- There was only a single error thrown for any issue pertaining to serial ports, and it was often misleading.
    - This error was split into multiple and additional data was included in the error reports.
    - https://github.com/esp-rs/espflash/pull/139
- macOS Monterey and/or systems with M1 processors were not having _any_ serial devices enumerated.
    - This has been fixed in the upstream `serialport` library, and `espflash` now uses the most recent version, so this issue should be resolved.
- On some or all installations of macOS, the `/dev/tty.*` ports are listed but the `/dev/cu.*` ports are not; because of this an error may mistakenly be returned indicating that the port does not exist.
    - The workaround for this is of course to use the `/dev/tty.*` variant of the port. I have not yet ascertained whether or not we will be able to get `list_available_ports()` to include the `/dev/cu.*` variants. However, since there is a seemingly suitable workaround I'm not considering this a major issue at this point in time.

stevecheckoway, 2022-04-13:
Thanks for getting this working! I'll test it out as soon as I'm able, but it sounds like the issues I was encountering have been resolved.

> * The workaround for this is of course to use the `/dev/tty.*` variant of the port. I have not yet ascertained whether or not we will be able to get `list_available_ports()` to include the [...]

I would be inclined to not bother with using the `/dev/cu.*` ports ever. They're really only useful for talking to a modem (the Calling Unit) where you can have `getty` listening on the `/dev/tty.*` port for call ins and still use the `/dev/cu.*` to make calls. I have never needed to use the `/dev/cu.*` ports.

I've been using `/dev/tty.*` with `esptool.py` as well as in my own Rust code using the `serialport` crate to talk to my ESP8266, ESP32, ESP32-S2, ESP32-S3, and ESP32-C3 on macOS Monterey on an M1 for several months.

But that said, I don't think there's any harm in using them; it's just unnecessary.

Verequies, 2022-04-26:
espflash 1.3.0+ doesn't not work under WSL with usbipd
Connecting to the device over uart with the usual Python IDF utilities works fine.

Narrowing down the issue, it appears it started with commit https://github.com/esp-rs/espflash/commit/85c8eb84db7f0a534d73772069f864475ff4dfe6 .

The error shows as below:
```
Error: espflash::serial_not_found

  × The serial port '/dev/ttyUSB0' could not be found
  help: Make sure the correct device is connected to the host system

```

However the device is definitely connected and is listed under `/dev/`, as well as works with the Python IDF utilities.

espflash 1.4.1 works fine under a Linux host, and on Windows itself as expected.

jessebraham, 2022-05-10:
The WSL usbipd problems are being tracked here: https://github.com/esp-rs/espflash/issues/166

I'm going to close this issue for now, as it's gotten a little long-winded and there are multiple problems reported, most of which I hope have been resolved to some extent. If anybody continues to experience any of the issues mentioned here I would ask that you either create an issue, or comment on an existing issue if one exists.

If anybody feels this should be re-opened, please feel free to say such and/or re-open the issue.

jessebraham, 2022-05-18:
For anybody following this who would like to use callout ports (`/dev/.cu*`) please check out this issue to follow the progress on that front: https://github.com/esp-rs/espflash/issues/181