esp-rs/espflash #143: CTRL-C doesn't work anymore (Windows 11)

bjoernQ, 2022-03-14:
On master ( `2e9347d34d41d9e59b62a2293bae41a7c193bb27` ) CTRL-C doesn't work anymore (Windows 11 at least) when running with `--monitor`

Need to check when exactly that broke - not tested on other platforms yet

bjoernQ, 2022-03-14:
Worked at least on commit `eeb0e8f95d184209cc42be9c4fec15c7172729fc`

jessebraham, 2022-03-14:
[This line](https://github.com/esp-rs/espflash/commit/b720bd4fb2daf6dbb459b4c348dbc633305c6cca#diff-fa60d5f53521a691b0a1419c6b80c9fdcbf2291a1e19fe9345c112bc6555ac14R107) is the only real change I see to the monitor since the working commit hash you've posted. Sorry to be a bother, would you be able to confirm that this is the commit which caused the problem please?

bjoernQ, 2022-03-14:
Sorry I wanted to look into this already by myself but got distracted by some linker issues ....

However, I found something!

The root cause seems to be mentioned in https://github.com/crossterm-rs/crossterm/pull/629

If I change https://github.com/esp-rs/espflash/blob/2e9347d34d41d9e59b62a2293bae41a7c193bb27/espflash/src/cli/monitor.rs#L123 to `KeyCode::Char('C') => break,` it works again - but certainly would break things on other platforms.

So, we could wait for the next `crossterm` release or add a temporary fix by adding the uppercase cases there, too. I personally can live with this since I use a locally compiled `espflash` where I can patch this but probably others will run into that issue in the meantime

I have no idea when to expect the next release of `crossterm` containing the fix. (I tried it with the main branch and then it works without changes on our side)

bjoernQ, 2022-03-14:
That should resolve it without breaking MacOS or Linux. And even when _crossterm_ gets the fix it should still work.

closed 2022-03-14 by bjoernQ

WindSoilder, 2022-03-21:
Hi, Sorry for introcing a bug like this in crossterm, for now the new version 0.23.1 is fixed.

PR #145 (2022-03-14): Apply a temporary fix to the serial monitor to make Ctrl-c work again
Fixes #143