esp-rs/espflash #146: Autodetect flash size and speed and configure automatically

MabezDev, 2022-03-15:
Building on top of https://github.com/esp-rs/espflash/pull/141, we could detect and automatically update the image header.

JurajSadel, 2022-04-28:
This should be already implemented.

JuliDi, 2022-12-11:
There seems to be an issue with this detection (in version 1.7.0 at least).

If I use `espflash` directly, it correctly determines the flash size of 4 MB and the program, which is around 1 MB, uses up around 25% of the flash.
```bash
$ cargo run                                    
    Finished dev [optimized + debuginfo] target(s) in 0.15s
     Running `espflash --monitor target/riscv32imc-esp-espidf/debug/esp_pid`
New version of espflash is available: v2.0.0-rc.2

Detected 2 serial ports. Ports which match a known common dev board are highlighted.

Serial port: /dev/cu.usbmodem4201
Connecting...

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       34:b4:72:80:1d:e0
App/part. size:    992624/4128768 bytes, 24.04%
```

If I use `cargo espflash` on the same device however, it determines the flash size correctly but then my almost 1 MB program uses up 95% of the flash. If I add some more std bloat, I easily get beyond 1 MB and will refuse to flash.

```bash
cargo espflash --monitor /dev/cu.usbmodem4201                   
New version of cargo-espflash is available: v2.0.0-rc.2

Serial port: /dev/cu.usbmodem4201
Connecting...

    Finished dev [optimized + debuginfo] target(s) in 0.61s
Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       34:b4:72:80:1d:e0
App/part. size:    992624/1048576 bytes, 94.66%
```

eddwinpaz, 2024-03-04:
> There seems to be an issue with this detection (in version 1.7.0 at least).
> 
> If I use `espflash` directly, it correctly determines the flash size of 4 MB and the program, which is around 1 MB, uses up around 25% of the flash.
> 
> ```shell
> $ cargo run                                    
>     Finished dev [optimized + debuginfo] target(s) in 0.15s
>      Running `espflash --monitor target/riscv32imc-esp-espidf/debug/esp_pid`
> New version of espflash is available: v2.0.0-rc.2
> 
> Detected 2 serial ports. Ports which match a known common dev board are highlighted.
> 
> Serial port: /dev/cu.usbmodem4201
> Connecting...
> 
> Chip type:         ESP32-C3 (revision 3)
> Crystal frequency: 40MHz
> Flash size:        4MB
> Features:          WiFi
> MAC address:       34:b4:72:80:1d:e0
> App/part. size:    992624/4128768 bytes, 24.04%
> ```
> 
> If I use `cargo espflash` on the same device however, it determines the flash size correctly but then my almost 1 MB program uses up 95% of the flash. If I add some more std bloat, I easily get beyond 1 MB and will refuse to flash.
> 
> ```shell
> cargo espflash --monitor /dev/cu.usbmodem4201                   
> New version of cargo-espflash is available: v2.0.0-rc.2
> 
> Serial port: /dev/cu.usbmodem4201
> Connecting...
> 
>     Finished dev [optimized + debuginfo] target(s) in 0.61s
> Chip type:         ESP32-C3 (revision 3)
> Crystal frequency: 40MHz
> Flash size:        4MB
> Features:          WiFi
> MAC address:       34:b4:72:80:1d:e0
> App/part. size:    992624/1048576 bytes, 94.66%
> ```

If I try, 
```
cargo run
```
it runs the application with no issues. but when I try to flash it using
```
cargo espflash flash
```
it blames that the espflash image is too big.

SergioGasquez, 2024-03-04:
When using `cargo-espflash` you use the esp-idf partition table, while when using `espflash`, you use the `espflash` generated partition table, see: https://github.com/esp-rs/espflash/tree/main/cargo-espflash#bootloader-and-partition-table