esp-rs/espflash #153: Windows/JTAG-SERIAL Issues

bjoernQ, 2022-04-07:
Flashing on Windows (11) via JTAG-SERIAL has some issues
- the chip is not reset after flashing
- monitoring doesn't start after flashing if invoked with the `--monitor` option (most probably because it "hangs" in an attempt to reset the chip?)

Windows Version: Microsoft Windows [Version 10.0.22000.556]

Used the ESP-Rust board, also a vanilla ESP32C3 shows the same behavior

MabezDev, 2022-04-11:
@JurajSadel I believe you ran into this at one point? Any ideas?

bjoernQ, 2022-04-11:
I can make it locally work for me with replacing `reset` in `connection.rs` with this
```rust
pub fn reset(&mut self) -> Result<(), Error> {
        sleep(Duration::from_millis(100));

        if self.port_info.pid == USB_SERIAL_JTAG_PID {
            self.serial.write_data_terminal_ready(false)?;
            self.serial.write_request_to_send(false)?;

            sleep(Duration::from_millis(100));

            self.serial.write_data_terminal_ready(false)?;
            self.serial.write_request_to_send(false)?;

            sleep(Duration::from_millis(100));

            self.serial.write_request_to_send(true)?;
            self.serial.write_data_terminal_ready(false)?;
            self.serial.write_request_to_send(true)?;

            sleep(Duration::from_millis(100));

            self.serial.write_data_terminal_ready(false)?;
            self.serial.write_request_to_send(false)?; 
    
        } else {
            self.serial.write_data_terminal_ready(false)?;
            self.serial.write_request_to_send(true)?;
    
            sleep(Duration::from_millis(100));
    
            self.serial.write_request_to_send(false)?;    
        }

        Ok(())
    }
``` 

Probably even less code would do ... just copied and adapted the code from `reset_to_flash` - CTRL+R still doesn't work then since there we have similar code duplicated (so changing it there should do the trick) - however I'm wondering why it's apparently just me and why it's only on Windows

JurajSadel, 2022-04-11:
@MabezDev For some reason, the `SW reset` or `write_requet_to_send()` is not working correctly on Windows, when the chip (with USB-serial-JTAG) is in download mode. Unfortunately, I don't know the root cause of this, yet - if it's in `serialport` or in a deeper layer like the driver, for example, and I'm not sure, how to debug it.  I'm sorry I didn't bring anything new to this issue.

@bjoernQ you're not alone, I do have the same issues on Windows.

PR #157 (2022-04-12): bugfix: fix reset sequence for usb-serial-jtag chips
Small refactor of @bjoernQ's workaround in #153. 

Closes #153
closed 2022-04-12 by MabezDev