esp-rs/espflash #154: Broken pipe error on ESP32-C3 for "large" applications

gyscos, 2022-04-08:
It seems that on my ESP32-C3 (The QT PY board from adafruit), after flashing an application in bootloader mode, flashing fails for "large" applications.

All these situations lead to a successful flash:
* Reset in bootloader mode (hold BOOT, press RESET), then `cargo espflash`
* Flash an application from the Arduino IDE (I suppose it includes its own bootloader), then `cargo espflash`
* Flash a "small" application (less than ~200 segments)

Here is an example of flashing a "larger application" (~500 segments), right after having flashed from Arduino IDE:
```
Serial port: /dev/ttyACM0
Connecting...

    Finished release [optimized] target(s) in 0.05s

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       10:91:a8:31:06:e4
[00:00:00] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:08] ########################################     489/489     segment 0x10000

Flashing has completed!
```

Then, running `cargo espflash` right after gives this output:

```
Serial port: /dev/ttyACM0
Connecting...

    Finished release [optimized] target(s) in 0.05s

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       10:91:a8:31:06:e4
[00:00:00] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
Error: espflash::serial_error

  × Communication error while flashing device
  ├─▶ IO error while using serial port: Broken pipe
  ╰─▶ Broken pipe
```
(Before the error message, I see segments being uploaded; it fails around segment 200.)

So I suspect the bootloader included in here is not quite the same as the one Arduino IDE includes; in particular, it doesn't seem to support well applications larger than 200 segments.

jessebraham, 2022-04-08:
Thanks for opening this issue, sorry to hear you're having troubles.

I took at a look at the [schematic for that board](https://cdn-learn.adafruit.com/assets/assets/000/109/793/original/adafruit_products_QTC3_sch.png?1647545127) and it seems they are using the built-in `USB_SERIAL_JTAG` peripheral. We are doing the same on the [esp-rust-board](https://github.com/esp-rs/esp-rust-board) and have had reports of similar behaviour when testing (see: https://github.com/esp-rs/espflash/issues/153). We're currently trying to evaluate where exactly the problem lies, so I will update this issue as any new information is available.

gyscos, 2022-04-08:
I tried to copy the bootloader that the Arduino IDE uses into espflash, to see if that solves my issue.
I used `~/Arduino/hardware/espressif/esp32/tools/sdk/esp32c3/bin/bootloader_dout_80m.bin`.

It did not: even after using this new patched espflash to flash an application (and the bootloader), subsequent flashes still failed around the ~150 chunks mark.

MabezDev, 2022-04-11:
@gyscos are you able to flash with esptool.py?

We've noticed some intermittent issues on some of [our boards](https://github.com/esp-rs/esp-rust-board) with a USB-C connector, its possible it could be a small hardware defect and the connector may need reflowing for example.

gyscos, 2022-04-11:
A) I did notice that if I touch both the board and the USB cable, I sometimes see the board rebooting, so it's indeed possible the USB connector is not perfectly soldered, or the cable itself has issues.
That being said, the errors or successes I am seeing are entirely reproducible; the "working" ways to flash it always work (hard-reset to bootloader, or flash after having flashed from Arduino IDE.)

I installed esptool.py and ran a few tests. It seems to behave neither exactly like `cargo espflash` nor like the arduino IDE (though I think the Arduino IDE uses esptool.py internally).

I can perform a few different actions:
* A: Try flashing a simple demo application from arduino IDE
* B: Hard reset to bootloader (hold BOOT, press RESET)
* C: Try running `cargo espflash --release` from a "larger" application (~500 chunks)
* E: Try running esptool from the same "larger" application
```
esptool.py --chip esp32c3 elf2image target/riscv32imc-esp-espidf/release/myapp
esptool.py --chip esp32c3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dout --flash_freq 80m --flash_size 4MB 0x10000 target/riscv32imc-esp-espidf/release/myapp.bin
```
* F: Any sequence of action leading to a flashing failure (as described in the first post).

(Note that operation A and E both probably use esptool internally, but work on different applications.)

The following sequences always work:
* Starting with `B`: BC, BA, BE
* A, C, E (, E, E, ...)
* A, E, E (, E, E, ...)
* F, A (Arduino IDE works after a failure)

The following sequences always fail:
* A, C, C -> F
* A, E, C -> F
* A, E, E, C -> F
* F, C
* F, E

MabezDev, 2022-04-26:
@gyscos Thanks for the super in-depth debugging! I think I've figured out what is going on, could you try out #169 and see if it fixes the issue for you?

If so, I'll clean it up and add a bit of explanation as to how it happens :).

jessebraham, 2022-04-27:
@gyscos Thank you so much for the detailed comment, it really helped us track down the problem!

closed 2022-04-27 by jessebraham

gyscos, 2022-04-27:
Glad I could be of help :)
And thanks for the amazing work!

PR #169 (2022-04-27): Hack to disable the rtc watchdog when connecting under the usb-serial-jtag
When the chip is reset via the USB-SERIAL-JTAG not everything is fully reset, the RTC domain stays active. Therefore, if the RTC watchdog is enabled before being put into download mode it will trigger when flashing, hence resetting the board and interrupting flashing.

The reason it always works on Arduino is that the RTC watchdog is explicitly disabled very early on in the boot process which doesn't seem to be the case for normal esp-idf applications, and therefore Rust applications. 

Closes #154