esp-rs/espflash #155: Add capability to initialize a spiffs partition

turgu1, 2022-04-10:
Hello,

It would be handy to have the capability of initializing the content of a spiffs partition using a mechanism similar to platformio: a `data` folder in the main project directory contains the files and folders to add to the spiffs partition. On request, it is packaged and transmitted to the ESP32 board in the partition requested by the user (or the first spiffs partition available).

Cheers and thanks to get Rust available for the ESP32ish devices!

Guy

igrr, 2022-04-10:
Hi @turgu1, the tool used for generating SPIFFS partitions used by esp-idf and by PlatformIO can be found here: https://github.com/espressif/esp-idf/blob/master/components/spiffs/spiffsgen.py. Implementing something similar in Rust is definitely possible, but would require someone to learn a few things about SPIFFS filesystem design — not impossible but might not be something that's going to happen quickly. In the meantime please try calling the above mentioned tool.

turgu1, 2022-04-10:
Thanks Yvan, I will do that. Cheers!

closed 2022-04-10 by turgu1