esp-rs/espflash #156: espflash does not take partitions.csv from project main folder

turgu1, 2022-04-10:
Hello,

I've modified the cargo.toml file adding the following at the end in three trials without success:

```
[package.metadata.espflash]
partition-table = "./partitions.csv"
```

or

```
[package.metadata.espflash]
partition_table = "./partitions.csv"
```
 or

```
[package.metadata.espflash]
partition_table = "partitions.csv"
```

The only way to get a specific partitions table instead of the default one is to use the `--partition-table` option. I would have expected the update to the cargo.toml file would do it.

MabezDev, 2022-04-11:
Are you confusing `Cargo.toml` with the [espflash specific configuration files](https://github.com/esp-rs/espflash/tree/master/espflash#configuration)?

turgu1, 2022-04-11:
From the Readme file https://github.com/esp-rs/espflash/tree/master/cargo-espflash it seems that the info must be placed into Cargo.toml

MabezDev, 2022-04-11:
Ah sorry, I was only reading the `espflash` README. Are you calling `cargo espflash` inside the project directory? It's also possible this might be failing in a project that uses cargo workspaces (judging by this code: https://github.com/esp-rs/espflash/blob/d1016ef4084d26ba440fdef63763df23699c65b8/cargo-espflash/src/main.rs#L111)

Any thoughts @jessebraham?

jessebraham, 2022-04-11:
Hmm that's strange, I will have to dig into this.

Just to be clear, are you receiving any error messages when trying to use your `paritions.csv` file or does it just silently fail and use the default partition table?

turgu1, 2022-04-11:
It is silent. No error message. I had to download the binary partition info from the ESP32 to find that it was not taking the table from the project.

turgu1, 2022-04-11:
I was thinking about the Cargo.toml of my project. Another aspect: why not take the `partitions.csv` file from the project's main folder if it exists instead of a default CSV file. Then no need to use the --partition-table option.

turgu1, 2022-04-11:
Other info: The problem seems to be the same using `$ espflash ...` or `$ cargo espflash ...`

jessebraham, 2022-04-11:
Looks like this has been broken for awhile, oops 😁 Sorry about that!

I was able to reproduce your problem, and using `cargo espflash` from this branch resolved it:  
https://github.com/jessebraham/espflash/tree/fixes/cargo-meta

Would you mind trying this branch out? If it fixes things I can get it merged and put out a patch release relatively quickly.

turgu1, 2022-04-11:
Hello!
Well... doesn't seems to work for me. In my Cargo.toml file the following defines the partitions.csv file at the end of the toml:

```
[package.metadata.espflash]
partition_table = "partitions.csv"
```

Here is the result of running the patched cargo-espflash:

```
turgu1@phobos:~/Dev/rust-trials/test$ cargo espflash
Error: cargo_espflash::partition_table_path

  × Specified partition table is not a csv file

```

Here is the content of the `partitions.csv` file:

```
# Name,   Type, SubType, Offset,   Size, Flags
nvs,      data, nvs,     0x9000,   0x5000,
otadata,  data, ota,     0xE000,   0x2000,
app0,     app,  ota_0,   0x10000,  0x1F0000,
app1,     app,  ota_1,   0x200000, 0x1F0000,
spiffs,   data, spiffs,  0x3F0000, 0x10000,
```

I'm on Linux so the end of lines is `\n` not `\r\n`

turgu1, 2022-04-11:
Here is the partitions.csv file
[partitions.csv](https://github.com/esp-rs/espflash/files/8468441/partitions.csv)

bjoernQ, 2022-04-13:
I tried Jesse's fix and it seems to work fine for me ... tested on a ESP32C3 (but that shouldn't make a difference).
Flashed with `cargo espflash` and this is the log from booting
```
I (31) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (31) boot: compile time 12:10:40
I (31) boot: chip revision: 3
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32c3: SPI Speed      : 80MHz
I (46) boot.esp32c3: SPI Mode       : DIO
I (51) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (79) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (87) boot:  2 app0             OTA app          00 10 00010000 001f0000
I (94) boot:  3 app1             OTA app          00 11 00200000 001f0000
I (102) boot:  4 spiffs           Unknown data     01 82 003f0000 00010000
I (109) boot: End of partition table
```

Looks good to me.

Maybe (since those things usually happen to me 😄 ): Did you really install the `cargo-espflash` from the branch? (i.e. checkout the branch and do `cargo install --path .` inside the `cargo-espflash` directory)

turgu1, 2022-04-13:
Hello bjoernQ, yes I did. Here is the sequence of commands that I have done:

```
turgu1@phobos:~/Dev/rust-trials/espflash/cargo-espflash$ cargo install --force --path .
  Installing cargo-espflash v1.5.0-dev (/home/turgu1/Dev/rust-trials/espflash/cargo-espflash)
    Updating crates.io index
  Downloaded miette-derive v4.4.0
  Downloaded miette v4.4.0
  Downloaded miniz_oxide v0.5.1
  Downloaded flate2 v1.0.23
  Downloaded libc v0.2.123
  Downloaded 5 crates (791.8 KB) in 0.72s
   Compiling libc v0.2.123
   Compiling proc-macro2 v1.0.37
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.91
   Compiling memchr v2.4.1
   Compiling autocfg v1.1.0
   Compiling cfg-if v1.0.0
   Compiling version_check v0.9.4
   Compiling serde_derive v1.0.136
   Compiling serde v1.0.136
   Compiling crc32fast v1.3.2
   Compiling adler v1.0.2
   Compiling typenum v1.15.0
   Compiling regex-syntax v0.6.25
   Compiling hashbrown v0.11.2
   Compiling smallvec v1.8.0
   Compiling log v0.4.16
   Compiling cpp_demangle v0.3.5
   Compiling unicode-width v0.1.9
   Compiling fallible-iterator v0.2.0
   Compiling signal-hook v0.3.13
   Compiling parking_lot_core v0.9.2
   Compiling pkg-config v0.3.25
   Compiling stable_deref_trait v1.2.0
   Compiling bitflags v1.3.2
   Compiling lazy_static v1.4.0
   Compiling maybe-uninit v2.0.0
   Compiling once_cell v1.10.0
   Compiling rustc-demangle v0.1.21
   Compiling cc v1.0.73
   Compiling scopeguard v1.1.0
   Compiling rustversion v1.0.6
   Compiling smawk v0.3.1
   Compiling ryu v1.0.9
   Compiling heck v0.4.0
   Compiling semver v1.0.7
   Compiling regex-automata v0.1.10
   Compiling serde_json v1.0.79
   Compiling camino v1.0.7
   Compiling remove_dir_all v0.5.3
   Compiling either v1.6.1
   Compiling nodrop v0.1.14
   Compiling fastrand v1.7.0
   Compiling is_ci v1.1.1
   Compiling cpufeatures v0.2.2
   Compiling itoa v0.4.8
   Compiling zero v0.1.2
   Compiling strsim v0.10.0
   Compiling termcolor v1.1.3
   Compiling pico-args v0.4.2
   Compiling zeroize v1.5.4
   Compiling owo-colors v3.3.0
   Compiling number_prefix v0.4.0
   Compiling itoa v1.0.1
   Compiling strum v0.24.0
   Compiling slip-codec v0.3.2
   Compiling md5 v0.7.0
   Compiling miniz_oxide v0.5.1
   Compiling generic-array v0.14.5
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling indexmap v1.8.1
   Compiling lock_api v0.4.7
   Compiling memoffset v0.6.5
   Compiling miniz_oxide v0.4.4
   Compiling libudev-sys v0.1.4
   Compiling array-init v0.0.4
   Compiling xmas-elf v0.8.0
   Compiling backtrace v0.3.64
   Compiling aho-corasick v0.7.18
   Compiling csv-core v0.1.10
   Compiling os_str_bytes v6.0.0
   Compiling quote v1.0.18
   Compiling smallvec v0.6.14
   Compiling atty v0.2.14
   Compiling flate2 v1.0.23
   Compiling signal-hook-registry v1.4.0
   Compiling terminal_size v0.1.17
   Compiling ioctl-rs v0.1.6
   Compiling termios v0.2.2
   Compiling serial-core v0.4.0
   Compiling mio v0.8.2
   Compiling tempfile v3.3.0
   Compiling dirs-sys-next v0.1.2
   Compiling regex v1.5.5
   Compiling nix v0.23.1
   Compiling gimli v0.26.1
   Compiling supports-color v1.3.0
   Compiling supports-unicode v1.0.2
   Compiling supports-hyperlinks v1.2.0
   Compiling object v0.27.1
   Compiling serial-unix v0.4.0
   Compiling block-buffer v0.10.2
   Compiling crypto-common v0.1.3
   Compiling libudev v0.3.0
   Compiling parking_lot v0.12.0
   Compiling directories-next v2.0.0
   Compiling console v0.15.0
   Compiling signal-hook-mio v0.2.3
   Compiling serial v0.4.0
   Compiling digest v0.10.3
   Compiling dialoguer v0.10.0
   Compiling indicatif v0.16.2
   Compiling crossterm v0.23.2
   Compiling sha2 v0.10.2
   Compiling serialport v4.1.0
   Compiling addr2line v0.17.0
   Compiling unicode-linebreak v0.1.2
   Compiling espmonitor v0.7.0
   Compiling thiserror-impl v1.0.30
   Compiling miette-derive v4.4.0
   Compiling bytemuck_derive v1.1.0
   Compiling binread_derive v2.1.0
   Compiling clap_derive v3.1.7
   Compiling strum_macros v0.24.0
   Compiling bytemuck v1.9.1
   Compiling thiserror v1.0.30
   Compiling textwrap v0.15.0
   Compiling miette v4.4.0
   Compiling binread v2.2.0
   Compiling clap v3.1.8
   Compiling bstr v0.2.17
   Compiling toml v0.5.8
   Compiling cargo-platform v0.1.2
   Compiling serde-hex v0.1.0
   Compiling serde_plain v1.0.0
   Compiling csv v1.1.6
   Compiling cargo_toml v0.11.5
   Compiling cargo_metadata v0.14.2
   Compiling espflash v1.5.0-dev (/home/turgu1/Dev/rust-trials/espflash/espflash)
   Compiling cargo-espflash v1.5.0-dev (/home/turgu1/Dev/rust-trials/espflash/cargo-espflash)
    Finished release [optimized] target(s) in 54.10s
   Replacing /home/turgu1/.cargo/bin/cargo-espflash
    Replaced package `unknown` with `cargo-espflash v1.5.0-dev (/home/turgu1/Dev/rust-trials/espflash/cargo-espflash)` (executable `cargo-espflash`)
turgu1@phobos:~/Dev/rust-trials/espflash/cargo-espflash$ cd ../../test
turgu1@phobos:~/Dev/rust-trials/test$ cargo espflash
Error: cargo_espflash::partition_table_path

  × Specified partition table is not a csv file

turgu1@phobos:~/Dev/rust-trials/test$ cargo espflash --partition-table partitions.csv
Error: cargo_espflash::partition_table_path

  × Specified partition table is not a csv file

turgu1@phobos:~/Dev/rust-trials/test$
```

turgu1, 2022-04-13:
Sorry Guys, I found the problem!!

When I cloned the GitHub project, I assumed that I was getting automatically into the proper branch. I was not :-(. 

I've changed to the proper branch and did a test again. This time it worked fine.

Very sorry! And thanks for your effort!!
Guy

jessebraham, 2022-04-13:
These things happen, I'd be lying if I said I hadn't done the same 😁 Thank you for confirming that this fix does indeed work, I'll get it merged today.

There are a couple other small bugs I'd like to look into first, but I will release `1.4.1` soon including this fix. I will leave this issue open until the new version has been published.

turgu1, 2022-04-13:
Thanks! One aspect that would be handy: as the option is named --partition-table, why not using also the same name in the Cargo.toml `[package.metadata.espflash]` portion:

Instead of:

```
[package.metadata.espflash]
partition_table = "partitions.csv"
```

or in addition to, using the following (using `-` instead of `_`):

```
[package.metadata.espflash]
partition-table = "partitions.csv"
```

Thanks again! Cheers to the team!
Guy

jessebraham, 2022-04-19:
I've released a new version of `cargo-espflash` and `espflash`, 1.4.1 ([release notes](https://github.com/esp-rs/espflash/releases/tag/v1.4.1)), which contains the fix for this issue.

closed 2022-04-19 by jessebraham

thomas725, 2023-01-29:
Is it possible that this functionality has been broken again in later versions?

I too have to call
```
espflash --monitor --partition-table partitions.csv target/xtensa-esp32-espidf/release/playground-2023-01-15
```
instead of just `cargo run` to make it use my `partitions.csv` definition, even though my `Cargo.toml` file contains those two lines:
```
[package.metadata.espflash]
partition_table = "partitions.csv"
```

See also:
https://gitlab.com/thomas351/esp32-rust-playground

surfingreg, 2024-12-29:
Exact same as @thomas725 (now December 2024, espflash version 3.2.0 on Linux Debian).

I can successfully call ```espflash flash --partition-table partitions.csv --monitor target/xtensa-esp32s3-espidf/release/esp-ota-std-ex``` to yield
```
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00200000
```
which reflects my partitions.csv: 
```
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M,
```

Whereas if I run ```cargo run --release``` with Cargo.toml
```
[package.metadata.espflash]
partition_table = "partitions.csv"
```
...I see what I believe is the default partition table.
```
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
```

I did confirm my ~/.config/espflash/espflash.toml takes precedence when running cargo but is overruled by the explicit --partition-table switch. In other words, Cargo.toml has no effect in any scenario I can find.

surfingreg, 2024-12-29:
It appears the current solution is ```.cargo/config.toml``` with:
```
[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor --partition-table=partitions.csv"
rustflags = [ "--cfg",  "espidf_time64"] 
```

Works fine via ```cargo run```.