esp-rs/espflash #166: espflash 1.3.0+ doesn't not work under WSL with usbipd

Verequies, 2022-04-26:
As the title suggests, commits post 1.2.0 break compatibility with WSL and usbipd.
Connecting to the device over uart with the usual Python IDF utilities works fine.

Narrowing down the issue, it appears it started with commit https://github.com/esp-rs/espflash/commit/85c8eb84db7f0a534d73772069f864475ff4dfe6 .

espflash 1.4.1 works fine under a Linux host, and on Windows itself as expected.

andywswan, 2022-05-05:
same here, wsl, espflash v1.4.1

minicom can see the board perfeclty

```
Compiled on Dec 23 2019, 02:06:26.
Port /dev/ttyUSB0, 13:35:04

Press CTRL-A Z for help on special keys

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xf (BROWNOUT_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x147c
load:0x403ce000,len:0x91c
load:0x403d0000,len:0x2838
entry 0x403ce000

################################################
arch:ESP32C3, 3
compile_time:Ai-Thinker|B&T
ble_mac:34B4724c46DA
wifi_mac:34B4724c46D8
sdk_version:v4.3-beta3-195-g6be10fab0
firmware_version:2.2.0
compile_time:Jul  5 2021 13:46:35

ready

################################################
```

jessebraham, 2022-05-07:
I don't use Windows, so unfortunately I can't make any suggestions. It's possible this may be an issue with the [serialport](https://github.com/jessebraham/serialport-rs) library, but again I have no way to verify.

KorvinSzanto, 2022-05-07:
I am also seeing this same behavior. `screen` can read from the serial port and I've had no trouble working on rp2040 boards through usbipd.

jessebraham, 2022-05-07:
@JurajSadel whenever you get a chance would you be able to take a look at this please?

KorvinSzanto, 2022-05-07:
> It's possible this may be an issue with the [serialport](https://github.com/jessebraham/serialport-rs) library

I tested this out and [I'm able to interact with `/dev/ttyUSB0` using the latest version of the serialport library without issue](https://gist.github.com/KorvinSzanto/83db81ab3198c2b4cee16ec4403fc998). Also worth noting the v1.2.0 release uses the same serialport version, though I wasn't able to get it compiling to test.

I'm not sure but judging by my output the problem is that the port type is considered "unknown" rather than "usb". Adding a flag to allow `SerialPortType::Unknown` types wherever `SerialPortType::UsbPort` is explicitly allowed might fix this.

andywswan, 2022-05-08:
> It's possible this may be an issue with the [serialport](https://github.com/jessebraham/serialport-rs) library

Kind of, but not quite. I think the true reason for this is that Ubuntu's usbip database of USB hardware identifiers does not have this kind of device (which is true for all new devices until the drivers update anyway), so the serialport crate honestly describes this port as Unknown. The question is - how should we react on that. Theoretically it should be enough to just add our port data into `~/.config/espflash/espflash.toml` like so

```
[connection]
serial = "/dev/ttyUSB0"

[[usb_device]]
vid = "1a86"
pid = "7523"
```

but for whatever reason it didn't work for me. The port still could not be found despite the crate reads the conf 100% (if I make an intentional mistake there - it would yell at me). I'm not saying it's a bug, I might have missed something from the description.

However this issue will end, the error message `The serial port '/dev/ttyUSB0' could not be found` is very misleading, that's why it took me that much time to figure it out. The port not "could not be found", it could be found perfectly and even communicated with, but its type is unknown because it is reported as such by another crate "serialport" so please add VID and PID manually.

andywswan, 2022-05-09:
guys, for those who need to work with espflash on wsl NOW (me) I've created a really quick fix here https://github.com/andywswan/espflash that works like so:

```
Running `target/debug/espflash board-info`

Ports found by serialport crate: [SerialPortInfo { port_name: "/dev/ttyUSB0", po
rt_type: Unknown }]

Port name from espflash.toml: /dev/ttyUSB0

Adding VID and PID from conf file: SerialPortInfo { port_name: "/dev/ttyUSB0", p
ort_type: UsbPort(UsbPortInfo { vid: 6790, pid: 29987, serial_number: None, manu
facturer: None, product: None }) }

Serial port: /dev/ttyUSB0
Connecting...

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       34:b4:72:4c:46:d8
```

it is not a solution to the problem though as it just gets port data from espflash.toml and ignores all the previous logic completely

JurajSadel, 2022-05-17:
I'm sorry for the delay, can you guys test #179 if it resolves issues with `wsl2`, please?
@Verequies, @andywswan,  @KorvinSzanto

PR #179 (2022-05-20): bugfix: flash with wsl2
Closes #166 

Note: It is not possible to flash `usb-serial-jtag` chips with `WSL2` because the reset also resets `serial-jtag-peripheral` which disconnects the chip from WSL2.
closed 2022-05-20 by jessebraham