esp-rs/espflash #171: Broken pipe while flashing over USB, on macOS only

allison-n-h, 2022-04-29:
Trying to test flashing an image using the USB port on an ESP32S3 DevKit-C-1-N8. For larger images (like the [wifi station example](https://github.com/espressif/esp-idf/tree/v4.4/examples/wifi/getting_started/station) from the idf), flashing fails about ~75% through the image:

```
Serial port: /dev/tty.usbmodem14101
Connecting...

Chip type:         ESP32-S3
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       7c:df:a1:e2:4a:a8
[00:00:00] ########################################      13/13      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
Error: espflash::serial_error

  × Communication error while flashing device
  ├─▶ IO error while using serial port: Broken pipe
  ╰─▶ Broken pipe
```

This seems to only be an issue with espflash on macOS and using the USB port on the devkit. Using the UART port works as expected, and using the native `idf.py flash` works as expected. It also seems to occur only 'larger' images - for example, flashing the hello_world example from the IDF completes successfully. Testing this on a Windows 11 machine, we don't see this issue either, so it does seem to be macOS specific.

We did verify we are using the 1.4.1 version of espflash.

MabezDev, 2022-04-29:
Can you try using master? We recently merged #169 which should fix this issue.

allison-n-h, 2022-04-29:
I seem to get a different failure when using the latest from master:

```
Connecting...

Chip type:         ESP32-S3
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       7c:df:a1:e0:31:3c
Error: espflash::timeout

  × Communication error while flashing device
  ╰─▶ Timeout while running WriteReg command

```

jessebraham, 2022-05-11:
I've unfortunately never seen this error in practice, so I'm a little stumped as to what the issue is. I run macOS as well and have never run into issues flashing the ESP32-S3, although I haven't used it much.

Just out of curiosity, which version of macOS are you running? Also if you're able to share your project (either a repo or a zip file, or whatever is easiest) that would be helpful so I could at least try to reproduce this.

allison-n-h, 2022-05-11:
We were seeing this on macOS Monterey 12.1, and trying to flash the app from this esp-idf example: https://github.com/espressif/esp-idf/tree/v4.4/examples/wifi/getting_started/station

MabezDev, 2022-05-11:
I can't seem to reproduce this.

- Running M1 Mac, Monterey 12.3. (also working on linux, just as another data point)
- Using espflash & cargo-espflash from current master.
- Can flash all examples from esp-hal
- Can also flash the station example listed above:
```bash
espflash --bootloader build/bootloader/bootloader.bin build/wifi_station.elf
Serial port: /dev/tty.usbserial-1110
Connecting...

Warning: could not detect flash size (FlashID=0x3980C2, SizeID=0x39), defaulting to 4MB

Chip type:         ESP32-S3
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       84:f7:03:80:09:64
[00:00:01] ########################################      13/13      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:37] ########################################     363/363     segment 0x10000

Flashing has completed!
```

Perhaps this is a hardware issue?

allison-n-h, 2022-05-11:
We tested on two different Intel Macs (one is a MBP 2019, the other is a MBP 2017), and encounter this issue.

mykmelez, 2022-05-20:
I wasn't able to reproduce this issue on my ESP32-S3-DevKitC-1-N8R2, so I picked up an ESP32-S3-DevKitC-1-N8 (the same devkit that @allison-n-h is using), and I'm able to reproduce the issue there on my M1 Mac with espflash 1.4.1:

```
Serial port: /dev/tty.usbmodem1101
Connecting...

Unable to connect, retrying with extra delay...
Unable to connect, retrying with default delay...

Chip type:         ESP32-S3
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       f4:12:fa:40:67:14
[00:00:00] ########################################      13/13      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
Error: espflash::serial_error

  × Communication error while flashing device
  ├─▶ IO error while using serial port: Broken pipe
  ╰─▶ Broken pipe
```

I can't reproduce it on espflash 1.5.1 or the current tip of the master branch (d5583ef1e67d025cd515913c0136e5e0f356637f), but that's only because I get a different error there (which is again the same one that @allison-n-h sees on those versions of espflash):

```
Serial port: /dev/tty.usbmodem1101
Connecting...

Chip type:         ESP32-S3
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       f4:12:fa:40:67:14
Error: espflash::timeout

  × Communication error while flashing device
  ╰─▶ Timeout while running WriteReg command
```

jessebraham, 2022-05-24:
Thank you for the additional information! This at least highlights some hardware differences which may be causing these issues. I only have the `ESP32-S3-DevKitC-1-N32R8V` presently, which may be why I'm unable to reproduce.

|           Board           |      Flash       |      PSRAM      | Status |
| :-----------------------: | :--------------: | :-------------: | :----: |
|   ESP32-S3-DevKitC-1-N8   |     8MB Quad     |     &mdash;     |   ❌   |
|  ESP32-S3-DevKitC-1-N8R2  |  8MB Quad  | 2MB Quad  |   ✅   |
| ESP32-S3-DevKitC-1-N32R8V | 32MB Octal | 8MB Octal |   ✅   |

Based on the above data it seems possible that somehow the lack of PSRAM is causing this. I will see if I can get some more hardware to verify this, though I'm not sure when I will receive it.

closed 2022-06-02 by MabezDev