esp-rs/espflash #173: Writing to flash memory data bus

CMuncey, 2022-05-01:
Can this program write to the data bus portion of flash memory on the esp32c3? I am currently trying to put my rodata section at 0x3C100000 (1MB past the start of flash, but in the data bus address space, to give plenty of space for my text section), but the data doesn't seem to be there when executing the code. The location is correct in the elf file, but when I read from 0x3C100000, it's just seemingly random data.

I somewhat fixed this by just placing my rodata at 0x42100000 (the instruction bus equivalent) and offsetting my addresses by 0x600000 (not that it really matters, since the instruction bus can be used for reading). This leads to very long flashing times for some reason however, I'm not sure why. It needs to flash two sections which I guess makes sense, but I'm not sure why it takes so much longer.

There looks to be a value specifying the start of drom in the source, but I'm not very familiar with rust so I can't tell how it's being used.

I'm not sure what all information might be useful, but I am not using any of the esp idf. I am using this to flash, the gnu risc-v toolchain to compile, a custom linker script, and writing everything in assembly as opposed to C or rust.

MabezDev, 2022-05-02:
If you are not using esp-idf, I think you'll be interested in [direct boot](https://github.com/espressif/esp32c3-direct-boot-example). Here is a [bare bones Rust demo](https://github.com/MabezDev/esp32c3-experiments) I wrote a while back too.

Once you have your application setup for direct boot, using `espflash`'s `direct-boot` argument should work.

CMuncey, 2022-05-02:
That's what I'm using, I have a linker script set up similar to the one on the espressif one and yours, but a little more simplified. Objdump tells me that the rodata section should be there, and it should have the data in the correct position, but when I read from that memory location it doesn't seem to be what I expect (just the lowercase alphabet, a newline, then a null character).

I haven't written a uart driver yet (and didn't know about the rom uart_tx function until looking at your code) so the only way I can test it at the moment is to load from the rodata address onto the stack and then use `fadd.s` to crash the program since it does not have hardware floating point support, and get the rom code's register and stack dump. Reading from rodata puts seemingly random data there, but manually placing the data there works fine.

CMuncey, 2022-05-02:
I think I fixed it? The linker script from the rust demo worked, so I played with mine until it worked too. It looks like the LMA needs to be an offset from 0 (which I guess makes sense, since 0x42000000/0x3C000000 is offset 0 from the start of flash), I was thinking that one also needed to be at the virtual address.

closed 2022-05-02 by CMuncey