esp-rs/espflash #175: Add support for building tests

svenstaro, 2022-05-09:
I think it would be pretty cool if espflash could build/merge/flash test binaries.

farbodpm, 2022-11-05:
there are some examples in probe-rs. have U checked it?

svenstaro, 2022-11-05:
I checked it but do not know exactly which files you mean.

dignifiedquire, 2023-10-28:
Any updates on this? would be great to use this to build test binaries to be executed in qemu and the end devices.

SergioGasquez, 2023-11-16:
> Any updates on this? would be great to use this to build test binaries to be executed in qemu and the end devices.

Not really, for [`esp-hal` we are using `probe-rs` and `teleprobe` for HIL](https://github.com/esp-rs/esp-hal/issues/809#issuecomment-1802434224). Currently it only supports RISC-V (except for C2,[which should be soon supported](https://github.com/probe-rs/probe-rs/pull/1869). Xtensa support for probe-rs is also WIP. I would recommend going for this approach

dignifiedquire, 2023-11-16:
> Xtensa support for probe-rs is also WIP. I would recommend going for this approach

is there a place to follow that progress?

SergioGasquez, 2023-11-16:
> is there a place to follow that progress?

I dont think there is any tracking issue, feel free to join the [matrix channel](https://matrix.to/#/#esp-rs:matrix.org) where we post updates or you can asks for them

bjoernQ, 2024-01-15:
There is a tracking issue for Xtensa in `probe-rs` now: https://github.com/probe-rs/probe-rs/issues/2001

SergioGasquez, 2024-07-18:
I think we can close this issue as we won't be adding such feature to espflash. `probe-rs` can already flash tests into all the targets but ESP32 (see https://github.com/probe-rs/probe-rs/issues/2354), for an example see https://github.com/esp-rs/esp-hal/tree/main/hil-test

closed 2024-07-18 by SergioGasquez