esp-rs/espflash #177: espflash does not follow symlinks on Linux

alxpettit, 2022-05-17:
```
$ espflash -V
espflash 1.5.1
$ uname -a
Linux ada-DT0 5.13.0-41-generic #46~20.04.1-Ubuntu SMP Wed Apr 20 13:16:21 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

In modern versions of Linux, the devfs system includes symlinks that allow you to unambiguously reference items by ID -- this includes serial devices.

e.g.
```
$ ls /dev/serial/by-id/ -lah
total 0
drwxr-xr-x 2 root root 80 May 16 19:55 .
drwxr-xr-x 4 root root 80 May 16 16:58 ..
lrwxrwxrwx 1 root root 13 May 16 19:19 usb-Raspberry_Pi_Picoprobe_[REDACTED]_if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 May 16 19:55 usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0
```

Instead of following symlinks passed for serial device file paths, it returns an error, like so:

```
$ espflash /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 target/xtensa-esp32-espidf/debug/rust-esp32-std-demo
Error: espflash::serial_not_found

  × The serial port '/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0' could not be found
  help: Make sure the correct device is connected to the host system
```

This leads me to use workarounds with `basename` and `readlink`, which definitely seems subpar and unnecessary. :P

MabezDev, 2022-05-17:
I've just opened #178, could you try it out and report back?

Seems to work for me

```rust
cargo run -p espflash -- --monitor /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_84:F7:03:41:79:AC-if00 target/riscv32imc-esp-espidf/debug/esp-fs-tests
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
     Running `target/debug/espflash --monitor '/dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_84:F7:03:41:79:AC-if00' target/riscv32imc-esp-espidf/debug/esp-fs-tests`
Serial port: /dev/ttyACM0
Connecting...

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       84:f7:03:41:79:ac
[00:00:00] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:09] ########################################     419/419     segment 0x10000

```

alxpettit, 2022-05-23:
Works great, thanks! :D