esp-rs/espflash #191: Implement support for Secure Boot V2

jessebraham, 2022-06-02:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html

farbodpm, 2022-10-25:
Hi what exactly here is needed ?

jessebraham, 2022-10-25:
I'm not entirely sure to be honest, I haven't read through the linked documentation yet. I believe it involves signing the image with some cryptographic key, there may be additional steps.

ostenning, 2024-05-27:
This is a pretty important production feature - whats the status of this issue is in the backlog? Any planned work to implement this?

jessebraham, 2024-05-27:
At this point we do not really have plans to implement any new features in `espflash`. While I understand this is important functionality, we unfortunately have not received consistent contributions from the community, and development/maintenance of these applications has been put mainly on my and @SergioGasquez's shoulders. I stopped maintaining this project largely for this reason. We both have other commitments at this time which are of higher priority, and `espflash` provides the required functionality for us to do our work at this time, so it's hard to justify spending any time here.

I would love to see a PR for this, I would be more than happy to review one. Otherwise, `esptool.py` is still a valid option, it can consume ELF files.

SergioGasquez, 2024-12-30:
I guess https://github.com/esp-rs/espflash/issues/715 and https://github.com/esp-rs/espflash/issues/713 are, somewhat, related to this issue

ivmarkov, 2025-01-13:
> I guess #715 and #713 are, somewhat, related to this issue

Yes, for secure boot, the only thing necessary is some way to add secure padding to the app image prior to calculating the sha256 signature.
At the moment I'm doing this with `espsecure.py`, but it would be great if `espflash` has the secure padding capability.