esp-rs/espflash #195: Error while running FlashDeflateBegin command

Evanfeenstra, 2022-06-13:
```
Error: 
  × The bootloader returned an error
  ├─▶ Error while running FlashDeflateBegin command
  ╰─▶ Bootloader failed to execute command
```

When I try to upload a release build. Is this because the binary is too big?

`save-image` gives me a file that is 2.3 MB. I am using `esp32-c3` with 4 MB flash. Is there a way to get more info about why it's not flashing?

When I remove some dependencies to get a smaller binary, it flashes just fine.

Evanfeenstra, 2022-06-15:
Ok trying it with esptool, the actual error is `boot: Image contains multiple DROM segments. Only the last one will be mapped.`

jessebraham, 2022-06-16:
Would you be able to share the the full commands with all options you ran to:

1. create the binary (ie. the invocation of the `save-image` subcommand)
2. flash said binary

I would like to try and reproduce this locally so I can hunt down the problem.

Evanfeenstra, 2022-06-16:
Heres a little repo that reproduces the problem: https://github.com/Evanfeenstra/esp32c3-test

It seems it was caused by a dependency on the secp256k1 crate that includes a bunch of C code. So I set CC to the riscv32-esp-elf-gcc included in the .embuild directory.

Interestingly, it works fine on ESP IDF 4.3.2... but not on 4.4 or mainline

BudiNverse, 2022-07-25:
Hi, I would like to report that I am also facing this issue on the `esp32`. Here is my setup and sample code.

**Board**: doit esp32 devkit v1
**OS**: Manjaro Linux x86_64 (Running on VMWare, Windows 10 Host)
**Kernel**: 5.13.19-2-MANJARO

```toml
# Cargo.toml

[package]
authors = ["zeon256"]
edition = "2021"
name = "esp32-sample"
resolver = "2"
version = "0.1.0"

[profile.release]
opt-level = "s"
# lto = "thin"

[profile.dev]
debug = true # Symbols are nice and they don't increase the size on Flash
opt-level = "z" 
# lto = "thin"

[features]
pio = ["esp-idf-sys/pio"]

[dependencies]
esp-idf-svc = {version = "0.42.1", features = ["experimental"] }
esp-idf-sys = {version = "0.31.6", features = ["binstart"]}
embedded-hal = "0.2"
# embedded-svc = {version = "0.22", features = ["experimental"]}
# esp-idf-hal = {version = "0.38", features = ["experimental"]}
log = "0.4"
anyhow = "1"

[build-dependencies]
anyhow = "1"
embuild = "0.29"

```

```rust
/// main.rs

use esp_idf_sys as _; // If using the `binstart` feature of `esp-idf-sys`, always keep this module imported
use anyhow::bail;
use log::info;
use std::{sync::Arc, time::Duration};

fn main() -> anyhow::Result<()> {
    esp_idf_svc::log::EspLogger::initialize_default();
    let delay = Duration::from_millis(500);
    loop {
        info!("hello");
        std::thread::sleep(delay);
    }
}
```

Env Var:
```
echo $LIBCLANG_PATH
/home/zeon256/Documents/xtensa-esp32-elf-clang/lib/libclang.so.13
```

I will give it another try on ubuntu and see if there is any difference

jessebraham, 2022-08-16:
Would either of you be able to try this again using the current `master` branch of the repository, please? I'm curious if #217 has resolved this or if there's something else going on.

Douile, 2022-09-01:
Did you try with `--use-stub` fixed write-bin-to-flash for me

elwerene, 2022-09-01:
> Did you try with `--use-stub` fixed write-bin-to-flash for me

My espflash does not have a `--use-stub`` option. I switched to esptool because of this error for now.

Tuupertunut, 2022-09-06:
I'm getting this error when trying to write binary data to my custom partition using `espflash write-bin-to-flash`.

jessebraham, 2023-01-25:
I've not experienced this error and there have been plenty of changes made since this issue was opened; due to the lack of activity I'm going to close this for these reasons.

If anybody feels this is the wrong action please feel free to re-open the issue or create a new issue, however please keep in mind that if I am unable to reproduce the problem then I am unable to debug and fix the problem.

closed 2023-01-25 by jessebraham

mykmelez, 2023-02-27:
FWIW, I can reproduce this issue with espflash 1.7.0 and an esp32-s3 with 8MiB flash, once my binary exceeds 2MiB; but I *can't* reproduce it after upgrading to espflash 2.0.0-rc.3. So whatever the issue was, it seems to have been resolved.