esp-rs/espflash #198: Add an option to flash file without having to build the project

SergioGasquez, 2022-06-23:
Maybe, if we have an argument pointing to an elf file, flash it instead of building the project and then flashing the resulting file.

MabezDev, 2022-06-23:
This is what `espflash` does already! Easy to confuse with `cargo-espflash` which is the cargo wrapper that also builds the project. They are actually two different executables.

SergioGasquez, 2022-06-27:
Sorry, I meant `cargo-espflash`. My point was to get this feature inside `cargo-espflash` so users that want to flash a file without building the project do not need to install `espflash` or rely on idf tool.

tcbennun, 2022-07-21:
Agreed on this. It actually can cause a rebuild in some cases (e.g. performing a workspace build then using `cargo espflash` to flash one of the binary crates).

MabezDev, 2022-07-28:
I believe this was closed via https://github.com/esp-rs/espflash/pull/205

jessebraham, 2022-08-09:
This was closed erroneously, re-opening.

jessebraham, 2022-09-15:
After thinking on this for awhile, I actually think it's not necessary. If the project is already built then normal execution is fine, the time added by `rustc` for an already built project is negligible.

closed 2022-09-15 by jessebraham