esp-rs/espflash #200: Flash timeout

farbodpm, 2022-06-25:
Hi, 
Espflash works fine when I am trying to program Default Uart pins on esp32s3 with cp2104 but When I use the Jtag I get timeout error. Does esp flash support this?

MabezDev, 2022-06-25:
Yes it does support usb serial JTAG. I use it everyday, so it _should_ work. Could you give us more details about your environment, package versions and finally the full output from espflash, thanks!

farbodpm, 2022-06-25:
Hardware : USB_N --> GPIO19
                  USB_P --> GPIO20
My Command: 
export ESP_ARCH=xtensa-esp32s3-espidf

espflash /dev/ttyACM0 target/${ESP_ARCH}/${BUILD_MODE}/ee 
                  

ERROR: 
Serial port: /dev/ttyACM0
Connecting...

^[Chip type:         ESP32-S3
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       f4:12:fa:44:a8:10
Error: espflash::timeout

  × Communication error while flashing device
  ╰─▶ Timeout while running WriteReg command

MabezDev, 2022-06-25:
Duplicate of https://github.com/esp-rs/espflash/issues/171. Please install espflash from master until the next release. `cargo install espflash --git https://github.com/esp-rs/espflash` should do the trick.

closed 2022-06-25 by MabezDev

farbodpm, 2022-06-26:
thanks for your response. It would be my pleasure to help to develop more feature on this great app.

Best Regards
@MabezDev