esp-rs/espflash #206: SHA-256 comparison failed: [...] attempting to boot anyway...

jessebraham, 2022-07-06:
When the app image header is updated with new flash configuration values, output like the following can be seen in the bootloader messages:

```text
SHA-256 comparison failed:
Calculated: bf6e7aa7738b60d6b15de2bea35979ca40a8a94224f3c78ef00f19a96becf599
Expected: c279089098b4d648c53538bb1dfba7caa16140817d4be0ee4394fe6967e6813d
Attempting to boot anyway...
```

While the application still executes as expected, we will need to address this. I have a local branch in-progress but it still needs some work.

SergioGasquez, 2024-01-31:
I did some investigations for this issue using an `esp-idf` example with `esptool` and `espflash`, here are my initial findings:
If we use: 
```
esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blink.bin

espflash monitor
```
Or the espflash equivalent:
```
espflash flash -M --port /dev/ttyACM0 -b 460800 --chip esp32c3 --flash-mode dio --flash-size 2mb --flash-freq 80mhz --bootloader build/bootloader/bootloader.bin --partition-table build/partition_table/partition-table.bin build/blink.elf
```
We don't get the comparison failed message. Note that we are using `--flash-mode dio --flash-size 2mb --flash-freq 80mhz ` in both commands.

From `esptool` docs:
[image]

In `espflash`, we are not using the same defaults [FlashSettings](https://github.com/esp-rs/espflash/blob/31817fa0112e23911aa0ccac840bf1f6b890a744/espflash/src/flasher/mod.rs#L261-L265), for example, for C3, we use 40MHz as default flash freq (while `esptool` default is 80MHz). If you try to use this freq in `esptool`:
```
esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blink.bin
```
We get the following warning:
```
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash frequency setting. Use the --flash_frequency=keep option instead of --flash_frequency=40m in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
```

Possible workarounds:
- Print a warning to the user when he is not using the defaults `FlashSettings`. Possibly also update default values to avoid the issue if we dont modify any `FlashSettings` param.
- Create a `--dont-append-digest` option for `save-image` arg and print a similar warning as `esptool`.

We can do one or both workarounds, any input or suggestions on how to proceed @esp-rs/espressif ?

Vollbrecht, 2024-01-31:
related info maybe -> general description from [esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html)

in esp-idf-sys we provide a macro to override the app_description via a weak link into the app binary [here](https://github.com/esp-rs/esp-idf-sys/blob/master/src/app_desc.rs), but on default its filled by the esp-idf build system

bjoernQ, 2024-01-31:
I _think_ I have something .... will create a PR soonish

PR #567 (2024-01-31): Re-calculate the bootloader's SHA256
Fixes #206 

We modify the ESP-IDF booloader's header - then we also need to re-calculate it's SHA256
closed 2024-01-31 by SergioGasquez