esp-rs/espflash #208: cargo espflash, ESP32-C3: connects before building, causing connection failure

tcbennun, 2022-07-21:
I haven't got to the bottom of this yet, but I've noticed that if my `cargo espflash` invocation results in a build, then my ESP32-C3 connection fails until I physically disconnect and reconnect it. I'm using the built-in USB interface on the ESP.

I think it's because a connection to the device is established, followed by a lengthy delay during build, at which point afterwards something times out on the ESP32, preventing further action.

MabezDev, 2022-07-22:
It _shouldn't_ timeout, as watchdogs are disabled whilst it waits in reset. To clarify, does it connect initially, reading back board info then fail once it tries to flash the app?

tcbennun, 2022-07-22:
Before the build, it successfully says:

```
Serial port: /dev/ttyACM1
Connecting...

WARN setting baud rate higher than 115200 can cause issues.
```

Then the build occurs, and once done, it fails to flash. When I unplug and plug back in, and re-run the command (this time of course, no build needs doing), it works fine.

I'll give a full log when I can, I can't seem to replicate it at the minute...

jessebraham, 2022-08-02:
Have you been able to replicate the problem, or can we close this issue?

tcbennun, 2022-08-08:
Sorry for the wait. I can't replicate it on my latest PCBs, which makes me think it was a hardware issue. I will close this.

I would like the option of being able to specify *not* to build the project before flashing (i.e. use existing build artifacts if available, else abort). Any objections to a PR for this?

closed 2022-08-08 by tcbennun

jessebraham, 2022-08-09:
PR would be welcome! We actually had an issue for this but it got closed by accident, so I've re-opened it:  
https://github.com/esp-rs/espflash/issues/198