esp-rs/espflash #215: support for esp32-pico devices

ryankurte, 2022-08-07:
hi folks, thanks for all your work on such excellent tooling!

i'm trying to use this with (known working with `esptool.py`) `ESP32-PICO` ICs, which results the error "Failed to connect to on-device flash" in `espflash::flash_connect`. afaik these are standard `ESP32`s (albeit with flash on chip instead of externally), so it should be possible to use these with `espflash` but may require some alternate configuration?

relates to: https://github.com/esp-rs/esp-idf-template/issues/48

ryankurte, 2022-08-07:
the docs suggest the configuration should be [the same](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/advanced-options.html#custom-spi-pin-configuration) for the d4 and v3, is it that this isn't supported by the ROM stub loader?

jessebraham, 2022-08-18:
Are you able to flash the device now that we're providing stubs, or is there still additional work required to close this?

jessebraham, 2022-09-14:
Going to close this now, feel free to re-open if there's anything left to do.

closed 2022-09-14 by jessebraham