esp-rs/espflash #221: cargo-espflash refuses to work when using `-Zbuild-std` instead of `cargo.toml` `unstable.build-std`

codysch, 2022-08-24:
I normally don't configure my `.config/cargo.toml` to do `-Zbuild-std` because I have a project where some targets are built with `build-std`, and some are not. Cargo's config doesn't provide for enabling `unstable.build-std` depending on target and/or cfg, so I pass it on the command line (via a helper script or directly).

cargo-espflash checks the `.config/cargo.toml` and rejects my configuration, blocking the build.

As a workaround, I've switched to either using a modified `cargo-espflash` with the `cargo_config.has_build_std()` commented out, or using `espflash` instead.

MabezDev, 2022-08-24:
Could you explain how you enable cargo features in Cargo.toml? I wasn't aware this was possible.

jessebraham, 2022-08-25:
I'm not able to reproduce this, it seems to be accepting the argument just fine for me.

To test I just commented out the `[unstable]` section in `esp32c3-hal`, which I confirmed resulted in build errors. However, subsequently running the following command succeeded and flashed the device:

```
cargo espflash --release --example=hello_rgb --features=smartled --monitor -Zbuild-std=core
```

jessebraham, 2022-09-09:
I don't believe there is actually an issue here (you've said you have `.config/cargo.toml`, when it should be `.cargo/config.toml`), so I'm going to close this. If you can provide a way to reproduce this issue then please feel free to re-open the issue.

codysch, 2022-09-16:
> Could you explain how you enable cargo unstable features in Cargo.toml? I wasn't aware this was possible.

> I don't believe there is actually an issue here (you've said you have `.config/cargo.toml`, when it should be `.cargo/config.toml`), so I'm going to close this. If you can provide a way to reproduce this issue then please feel free to re-open the issue.

That was just a typo, and was only included to describe the "normal" state of things. I've edited the description to use the right files.

This doesn't affect `esp32c3`, because the check in espflash singles out the xtensa architecture. So this won't show up on the riscv based chips.

Here's the diff I locally apply to espflash:

```
diff --git a/cargo-espflash/src/main.rs b/cargo-espflash/src/main.rs
index e57296b..55b5ee7 100644
--- a/cargo-espflash/src/main.rs
+++ b/cargo-espflash/src/main.rs
@@ -271,9 +271,9 @@ fn build(
     // cross-compilation for xtensa targets.
     // If it has not been set then we cannot build the
     // application.
-    if !cargo_config.has_build_std() && target.starts_with("xtensa-") {
-        return Err(Error::NoBuildStd.into());
-    };
 
     // Build the list of arguments to pass to 'cargo build'. We will always
     // explicitly state the target, as it must be provided as either a command-line

```

Reproduction steps:

1. `cargo generate https://github.com/esp-rs/esp-template -n espflash-broken-buildstd`
2. pick esp32s3
3. `cd espflash-broken-buildstd`
4. `sed -i -e 's/build-std.*//' .cargo/config.toml`
5. `cargo espflash  --monitor --release /dev/cu.usbserial-1413101 -Zbuild-std=core,alloc`

Output:
```
Serial port: /dev/cu.usbserial-1413101
Connecting...

Error: cargo_espflash::build_std

  × Failed to build project
  ╰─▶ 'build-std' not configured
  help: cargo currently requires the unstable 'build-std' feature, ensure that .cargo/config{.toml} has the appropriate options.
          	See: https://doc.rust-lang.org/cargo/reference/unstable.html#build-std
```

closed 2022-09-20 by jessebraham