esp-rs/espflash #222: save-image argument order doesn't work as a runner for cargo

Brad-Hesson, 2022-08-26:
The order of arguments for the save-image subcommand is `<chip> <source> <destination>`.  If it was `<chip> <destination> <source>`, then the command could be used as a runner for cargo.  This is because cargo appends the source file to the end of the given command.  Alternatively there could be flags for the arguments so that the ordering is user decidable: `-c <chip> -d <destination> -s <source>)`.

If there is another way for me to solve this without changing the cli then please let me know!  I do see how the current ordering makes sense aesthetically.

jessebraham, 2022-08-28:
Thanks for opening this issue. I agree, the current CLI leaves a lot to be desired, and it can definitely be improved. I have started work on redesigning the CLI for this application, which can be found in [this repo](https://github.com/jessebraham/espflash-cli-resdesign). I'm not sure when I will merge this changes, but I will adapt them to take this into consideration some time next week. I'll leave this issue open until those changes get merged.

Brad-Hesson, 2022-08-29:
Great, sounds good!  For anyone that comes across this issue, I used the [cargo-runner](https://crates.io/crates/cargo-runner) crate to make my work flow work in the meantime.

closed 2022-09-21 by jessebraham