esp-rs/espflash #224: RFC: Relicense espflash under Apache-2.0 and MIT license

tom-borcin, 2022-08-31:
Hello all,

firstly I want to emphesize that **this issue is an RFC**. Your comments are welcome.
I would like to open a discussion about the license we are using: GNU GENERAL PUBLIC LICENSE Version 2 (GPL-2.0).

### The problem

Espressif currently maintains 4 codebases for flashing of microcontrollers:

1. [Esptool](https://github.com/espressif/esptool) for flashing from OS
2. [esp-serial-flasher](https://github.com/espressif/esp-serial-flasher) for flashing from host microcontroller
3. [esptool-js](https://github.com/espressif/esptool-js) for flashing from web browser
4. [espflash](https://github.com/esp-rs/espflash) for flashing from OS

Maintaining 4 different codebases is difficult, costs a lot of resources, is prone to mistakes and defies DRY (Don't Repeat Yourself) principle. Our end goal is to have one common codebase that will be used by other three codebases. And we cose espflash as our common codebase.

Unfortunately, there is one problem that prevents us to use it as library whitin other projects: its [GPL-2.0 license](https://www.gnu.org/licenses/old-licenses/gpl-2.0.html).

[esp-serial-flasher](https://github.com/espressif/esp-serial-flasher) and [esptool-js](https://github.com/espressif/esptool-js) are both licensed under Apache 2.0, which is incompatible with GPL-2.0 due to the restriction that terminates the grant of patent rights if the license sues over patent infringement. You can read about it in more detail [here](https://www.apache.org/licenses/GPL-compatibility.html).

Any library licensed under GPL-2.0 will enforce this licensing on any program that uses it. You can read about it [here](https://www.gnu.org/licenses/old-licenses/gpl-2.0-faq.en.html#IfLibraryIsGPL). Because of this we would have to relicense [esp-serial-flasher](https://github.com/espressif/esp-serial-flasher) and [esptool-js](https://github.com/espressif/esptool-js) to GPL-2.0 as well as any SW that uses these codebases.

### Solution

From Espressif's and my perspective, the easiest and best solution to aforementioned problems would be relicensing of [espflash](https://github.com/esp-rs/espflash) to dual license it under MIT and Apache 2.0. Dual licensing under MIT and Apache 2.0 is widely accepted within Rust community and provides the most permissive licensing (MIT) that gives us patent grant (Apache 2.0) and as a whole is compatible with GPL-2.0.

### Open Discussion

On the other hand we may be wrong, so I want to spark a discussion. Let us know what you think about suggested solution and let us know if there are any other solutions you think should be considered. If you are a contributor and you agree with aforementioned solution to change current GPL-2.0 license to MIT and Apache 2.0, please state your consent in the comment.

CCing all espflash contributors: @icewind1991 @jessebraham @MabezDev @ryankurte @svenstaro @marcelbuesing @bugadani @JurajSadel @kelnos @tpambor @nim65s @bjoernQ @mertzt89 @pavlus @ducktec @ubergeek801 @alazartech @kyrias @maspetsberger @KerryRJ @zRedShift @SergioGasquez @sirhcel @mchodzikiewicz @hanhossain

MabezDev, 2022-08-31:
I agree with the motivation and consent to the relicensing.

zRedShift, 2022-08-31:
Same, I consent as well.

tpambor, 2022-08-31:
I am fine with relicensing.

jessebraham, 2022-08-31:
I fully support this change!

hanhossain, 2022-08-31:
I always wondered why it was GPL-2.0 anyway as most of the rust community is dual licensed MIT and Apache 2.0.

I consent to the relicensing.

svenstaro, 2022-08-31:
I agree to the relicensing.

mchodzikiewicz, 2022-08-31:
I also agree

mertzt89, 2022-08-31:
I'm in agreement

bjoernQ, 2022-08-31:
I fully agree with relicensing `espflash`.

SergioGasquez, 2022-08-31:
Agree!

kyrias, 2022-08-31:
I consent to the relicensing.

sirhcel, 2022-08-31:
I fully support relicensing. Thank you for bringing this up @tom-borcin!

marcelbuesing, 2022-08-31:
I consent to the relicensing.

alazartech, 2022-08-31:
I consent to the relicensing.

kelnos, 2022-08-31:
I consent to relicensing to dual MIT/Apache-2.0.

ducktec, 2022-08-31:
I consent to the proposed MIT/Apache-2.0 relicensing.

KerryRJ, 2022-09-01:
I too agree with the relicensing.

bugadani, 2022-09-01:
No objections from me

icewind1991, 2022-09-01:
I agree with relicensing my contributions to MIT/Apache-2.0

ubergeek801, 2022-09-01:
No argument here, and thanks for reaching out!

JurajSadel, 2022-09-02:
I agree with the relicensing.

pavlus, 2022-09-03:
No objections from my side.

maspetsberger, 2022-09-04:
I agree with the relicensing.

nim65s, 2022-09-04:
Agree

zRedShift, 2022-09-08:
@tom-borcin looks like all the contributors have agreed (@ryankurte reacted with a thumbs up, the rest in a comment), so I believe there's a unanimous green light to go ahead with the re-licensing.

jessebraham, 2022-09-08:
Thanks for the update @zRedShift! I believe Tomas has a couple days off, so either I or him will update this thread as soon as he's available.

Thank you everybody for playing nicely and agreeing to this! 😁 ❤️

Janrupf, 2022-09-08:
I agree with the relicensing.

jendakol, 2022-09-12:
I agree with the relicensing.

ryankurte, 2022-09-13:
fine by me

tom-borcin, 2022-09-13:
Thank you all for your comments. We have unanimous agreement to relicense espflash from GPL-2.0 to dual license of MIT and Apache-2.0. I will move on with relicensing.

jessebraham, 2022-09-14:
Closed by #235

closed 2022-09-14 by jessebraham