esp-rs/espflash #265: wsl1 ubuntu20.04.5 Error: espflash::serial_not_found

HoGC, 2022-10-19:
I can't flash with ubuntu20.04, but idf.py flash is working normally.
```
: espflash /dev/ttyS7 target/riscv32imc-esp-espidf/debug/esp-rust
Error: espflash::serial_not_found

  × The serial port '/dev/ttyS7' could not be found
  help: Make sure the correct device is connected to the host system
```

jessebraham, 2022-10-19:
@JurajSadel would you mind looking into this?

JurajSadel, 2022-10-26:
Hi @HoGC! Sorry for the late reply. 
What version and what command are you using, please?

HoGC, 2022-10-26:
esp-idf(v4.4.2) commit： `1b16ef6cfc2479a08136782f9dc57effefa86f661`
Create project: `cargo generate https://github.com/esp-rs/esp-idf-template cargo`
```
cargo generate https://github.com/esp-rs/esp-idf-template cargo
🤷   Project Name : rs-demo
🔧   Destination: /home/g/00build-tools/01esp/rs-demo ...
🔧   Generating template ...
✔ 🤷   ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, mainline = UNSTABLE) · v4.4
? 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? (beware: Dev Containers not avai✔ 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? (beware: Dev Containers not available for esp-idf v4.3.2) · false
✔ 🤷   STD support · true
✔ 🤷   MCU · esp32
```
build: `cargo build`
flash: `espflash /dev/ttyS3 target/xtensa-esp32-espidf/debug/rs-demo`
```
Error: espflash::serial_not_found

  × The serial port '/dev/ttyS3' could not be found
  help: Make sure the correct device is connected to the host system
```
but esptool.py is working normally.
read chip id: `esptool.py -p /dev/ttyS3 chip_id`
```
esptool.py v3.3.2-dev
Serial port /dev/ttyS3
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:88:dc:8c
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 3c:71:bf:88:dc:8c
Hard resetting via RTS pin...
```

JurajSadel, 2022-10-27:
@HoGC Thank you for providing more information. I followed your steps and I can't reproduce the issue, however, I'm using `WSL2` instead of `WSL1`.

HoGC, 2022-10-27:
@JurajSadel 
Can you install a new LTS, change wsl to 1, and then test it?
If you are using Ubuntu-20.04, you can install Ubuntu-22.04.
Use `wsl --set-version Ubuntu-22.04 1` to change the version

jessebraham, 2022-10-27:
Would you mind running the commands below from within WSL1 (making sure that the target device is connected!) and posting the output please? I just want to make sure this isn't an issue with the underlying `serialport` library.

```bash
$ sudo apt install pkg-config libudev-dev
$ git clone https://github.com/serialport/serialport-rs
$ cargo run --example=list_ports
```