esp-rs/espflash #268: failed to select a version for the requirement `textwrap = "^0.15.1"`

coltCN, 2022-10-24:
cargo install cargo-espflash failed

Caused by:
  failed to select a version for the requirement `textwrap = "^0.15.1"`
  candidate versions found which didn't match: 0.16.0, 0.15.0, 0.14.2, ...
  location searched: `tuna` index (which is replacing registry `crates-io`)
  required by package `clap v3.2.22`
      ... which satisfies dependency `clap = "^3.2.22"` of package `cargo-espflash v1.7.0`
  perhaps a crate was updated and forgotten to be re-vendored?

textwrap@0.15.1  was yanked

jessebraham, 2022-10-24:
Thanks for bringing this to my attention.

In the meantime you can use `cargo install --locked cargo-espflash` and it should still succeed. I hopefully will get a new release out soon.

jessebraham, 2022-10-24:
This should be resolved now (see https://github.com/mgeisler/textwrap/issues/484#issuecomment-1289277367).

closed 2022-10-24 by jessebraham