esp-rs/espflash #274: Allow retrieving progress in write_bin_to_flash()

svenstaro, 2022-10-26:
Currently `write_bin_to_flash()` does all its own printing and the progress is internal. I use `espflash` as a library with my own flashing application on top of it and it's currently impossible to retrieve the current flash progress from it. Could you perhaps change the function to take a `FnMut` that could be used to set the progress if desired?

jessebraham, 2022-10-31:
Thanks for opening this issue, this was actually a pretty major oversight on my behalf and should be dealt with prior to the `v2.0.0` release. Really `indicatif` should be gated behind the `cli` feature, and shouldn't be pulled in at all if being used as a library.

I'm happy to do the leg work to get this done, but if you don't mind how would you like to see the API looking for this exactly? I'm not entirely sure how to go about this, as while I can thread through a callback to _set_ the progress, don't know how to get the total size out as well, unless the callback just takes both values each invocation. The total number of chunks is calculated in the same function as the segment is written in (`write_segment`).

svenstaro, 2022-11-01:
I don't really see an issue with giving the callback the total size as well.

jessebraham, 2022-11-01:
Okay I'll do that then, just wanted to make sure that would work on your end. Thanks!

PR #283 (2022-11-07): Only display progress bars when `cli` feature is enabled, allow passing progress callback
Mostly just decoupling the progress bars from the flash targets in this. I think now all of the CLI-only dependencies are optional. Not 100% happy with it, but frankly it's good enough.

Closes #274
closed 2022-11-07 by jessebraham