esp-rs/espflash #285: espflash doesn't detect serial anymore

mbuesch, 2022-11-05:
Since recently espflash doesn't see the serial of my esp32 anymore:

```
$ cargo espflash  /dev/ttyUSB0 
Error: espflash::serial_not_found

  × The serial port '/dev/ttyUSB0' could not be found
  help: Make sure the correct device is connected to the host system

```

This is because `serialport` detects it as `SerialPortType::PciPort`.
I'm not so sure why it does this and what changed. It might be related to a Linux kernel update or some other system package update.

With the following patch applied espflash works again for me:

```
diff --git a/espflash/src/cli/mod.rs b/espflash/src/cli/mod.rs
index b89c828..3015fc9 100644
--- a/espflash/src/cli/mod.rs
+++ b/espflash/src/cli/mod.rs
@@ -166,7 +166,7 @@ pub fn connect(args: &ConnectArgs, config: &Config) -> Result<Flasher> {
     //       serial ports, we can just pretend these types don't exist here.
     let port_info = match port_info.port_type {
         SerialPortType::UsbPort(info) => info,
-        SerialPortType::Unknown => {
+        SerialPortType::Unknown | SerialPortType::PciPort => {
             debug!("Matched `SerialPortType::Unknown`");
             UsbPortInfo {
                 vid: 0,
diff --git a/espflash/src/cli/serial.rs b/espflash/src/cli/serial.rs
index dc52dab..c58da33 100644
--- a/espflash/src/cli/serial.rs
+++ b/espflash/src/cli/serial.rs
@@ -143,16 +143,19 @@ fn detect_usb_serial_ports() -> Result<Vec<SerialPortInfo>> {
 #[cfg(not(all(target_os = "linux", target_env = "musl")))]
 fn detect_usb_serial_ports() -> Result<Vec<SerialPortInfo>> {
     let ports = available_ports().into_diagnostic()?;
+
+println!("FOUND PTS {:?}", ports);
     let ports = ports
         .into_iter()
         .filter(|port_info| {
             matches!(
                 &port_info.port_type,
-                SerialPortType::UsbPort(..) | SerialPortType::Unknown
+                SerialPortType::UsbPort(..) | SerialPortType::PciPort | SerialPortType::Unknown
             )
         })
         .collect::<Vec<_>>();
 
+println!("FILT PTS {:?}", ports);
     Ok(ports)
 }
```

This change in behavior is not caused by a change in `espflash` or `serialport`, because an `espflash` binary that used to work before stopped working.
But I'm unsure where the real problem is and whether for example `serialport` depends on some behavior that wasn't really guaranteed by the system.

I am using a Debian Sid x64 system with a Linux kernel 6.0.

jessebraham, 2022-11-14:
I'm a little fuzzy on how it actually works, but I believe Debian maintains some form of "database" containing this information which `serialport` then reads from; I could be completely wrong about this for the record, but for whatever reason this is what is in my head.

I'm not against adding the `PciPort` variant to the match statements as you've done in your patch, to be honest they were only filtered out before because I'd never seen a device enumerated like this. If you'd like to PR the changes that'd be great, otherwise I will do so sometime this week.

mbuesch, 2022-11-14:
Thanks for your help.
The problem has since been fixed on my system. Whatever broke it (I still don't know what it was) has been updated.

However, I agree that we probably shouldn't filter this. I don't really see a reason. Also it gives the user the very confusing error message that it couldn't find the device /dev/ttyUSBX that the user just passed to the program and that the user knows does exist for sure.

I will prepare a PR to make this more robust, as done in the patch.

closed 2022-11-15 by jessebraham

faern, 2023-01-06:
Did anyone figure out what made the ESP chips/dev boards suddenly register as `PciPort`? Because I also recently ran into this issue. Upgrading to `espflash` 2.0.0-rc.2 only partially solves the problem. Yes, now `espflash` can detect the serial connection, but it fails to communicate over it. I know the USB data lines on my board are a bit faulty, so the issue is there. But the point is that it worked before this `PciPort` thing happened. So **if there is a way to make Linux treat the USB device as earlier** that would save me from this not-able-to-flash situation.

mbuesch, 2023-01-06:
No, I still don't know what happened back then. I think it was related to some Debian package update, because that was the only thing that changed.

faern, 2023-01-06:
Ok. I'm currently in a position where I'm not able to flash my custom PCBs at all any longer :/

shanemmattner, 2023-01-07:
I ran into the same issue on Fedora 36:

```
[shane@home i2c-test]$ espflash /dev/ttyACM0 target/riscv32imc-esp-espidf/debug/i2c-test
New version of espflash is available: v2.0.0-rc.2

Error: espflash::serial_not_found

  × The serial port '/dev/ttyACM0' could not be found
  help: Make sure the correct device is connected to the host system
```

When I use the latest binary release I get the same issue reported by @faern :

```
[shane@home i2c-test]$ ~/Downloads/cargo-espflash-x86_64-unknown-linux-gnu/cargo-espflash espflash flash -p /dev/ttyACM0 --bin ~/Desktop/Rust/Projects/esp32_rust/i2c-test/target/riscv32imc-esp-espidf/debug/i2c-test
[2023-01-07T18:51:27Z INFO ] Serial port: '/dev/ttyACM0'
[2023-01-07T18:51:27Z INFO ] Connecting...
[2023-01-07T18:51:28Z INFO ] Unable to connect, retrying with extra delay...
[2023-01-07T18:51:30Z INFO ] Unable to connect, retrying with default delay...
[2023-01-07T18:51:31Z INFO ] Unable to connect, retrying with extra delay...
[2023-01-07T18:51:33Z INFO ] Unable to connect, retrying with default delay...
[2023-01-07T18:51:35Z INFO ] Unable to connect, retrying with extra delay...
[2023-01-07T18:51:36Z INFO ] Unable to connect, retrying with default delay...
[2023-01-07T18:51:38Z INFO ] Unable to connect, retrying with extra delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

faern, 2023-01-07:
@shanemmattner Awesome! You might want to fix your link in the forum post, it's broken. Also maybe mention what we know: That the kernel/udev/whatever suddenly started classifying the USB device as `ID_BUS=pci` where it was previously `usb` or unknown (not sure which)

shanemmattner, 2023-01-07:
@faern thanks for adding info to the Fedora Forum discussion.

Upgrading to Fedora 37 fixes the issue.  I posted the `inxi` file, perhaps that will help narrow down the problematic update?

commit 3b177925 (2022-11-15): Allow SerialPortType::PciPort during port detection

The USB serial port might have been misdetected by the system
as PCI port (permanent port).

Fixes #285