esp-rs/espflash #289: v2.0.0.rc1 Doesn't support specifying baud-rate?

bsodmike, 2022-11-11:
Previously this was possible, `espflash --monitor blinky --speed 920000` but now, it only allows a single command?

Please clarify, thanks!

liebman, 2022-11-12:
The option changed to 
```
-b, --baud <BAUD>
```

bsodmike, 2022-11-12:
Trying `-h` does not show any of the options though. Only the Command listing?

jessebraham, 2022-11-12:
Flashing is now under a subcommand, and as mentioned `--speed` is now `--baud` (to be more in line with [esptool](https://github.com/espressif/esptool).

Running `espflash --help` will display the subcommands, running `espflash flash --help` will show the options for the `flash` subcommand. This is how [clap](https://docs.rs/clap/latest/clap/) works.

closed 2022-11-12 by jessebraham

bsodmike, 2022-11-12:
Thanks @jessebraham

andywwright, 2023-05-07:
not being able to see all the options is very confusing at first and very inconvenient then. in this particular "feature" the new version to me is a regression, clap or not.