esp-rs/espflash #290: Stacktrace / Automatic Address Decoding

mrchantey, 2022-11-11:
When we get errors they are encoded, ie:
```
0x40384cd6 - taskSelectHighestPriorityTaskSMP
    at C:\temp\idf\riscv32imc-esp-espidf\release\build\esp-idf-sys-ee2e774c5bbb1a7b\out\build\C:/temp/.embuild/espressif/esp-idf/release-v4.4/components/freertos\tasks.c:3376
3fc8b8c0: 0x00000000 0x00000000 0x00000000 0x3fc8f168 0x00000000 0x00000000 0x00000000 0x00000001
3fc8b8e0: 0x00001881 0x80000003 0x00000000 0x40383776 0x4038020a 0x00000000 0x00000000 0x403801ee
```
Is there a way to decode these using espflash? Back in the day I used the [arduino tool](https://github.com/me-no-dev/EspExceptionDecoder) but that doesnt support ESP32-C3 and I'm not sure how to hook it into the rust workflow.

I'm not very familiar with the vanilla [idf monitor tooling](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html#automatic-address-decoding), but would also be interested if there was a way to somehow call the `addr2line_ ` in a rusty way.

MabezDev, 2022-11-11:
This should be enabled by default, as long as you're using the `--monitor` flag.

mrchantey, 2022-11-11:
Oh interesting, ill have a go at isolating the issue.

mrchantey, 2022-11-11:
Ok I followed these steps:

1. [image]

2. ```
	// main.rs
	use esp_idf_sys as _;

	fn main() {
	    esp_idf_sys::link_patches();
	    println!("Hello, world!");
	    do_silly_thing();
	}

	fn do_silly_thing() {
	    panic!("that was silly.");
	}
	```

3. `cargo espflash COM3 --monitor`
4. Output:
	```
	Hello, world!
	thread 'main' panicked at 'that was silly.', src\main.rs:12:5
	note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace    
	abort() was called at PC 0x4202071d on core 0
	0x4202071d - panic_abort::__rust_start_panic::abort
	    at C:\Users\petey\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\panic_abort\src\lib.rs:42
	Core  0 register dump:
	MEPC    : 0x40380884  RA      : 0x4038386a  SP      : 0x3fc8fdc0  GP      : 0x3fc8a600
	0x40380884 - panic_abort
	    at C:\Users\petey\work-temp\idf-test\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-9e04e690dded12b8\out\build\C:/Users/petey/work-temp/idf-test/.embuild/espressif/esp-idf/release-v4.4/components/esp_system\panic.c:402
	0x4038386a - __ubsan_include
	    at C:\Users\petey\work-temp\idf-test\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-9e04e690dded12b8\out\build\C:/Users/petey/work-temp/idf-test/.embuild/espressif/esp-idf/release-v4.4/components/esp_system\ubsan.c:294
	TP      : 0x3fc73e9c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
	S0/FP   : 0x3fc8fdfc  S1      : 0x3fc8fdfc  A0      : 0x3fc8fdfc  A1      : 0x3fc8fdde
	A2      : 0x00000000  A3      : 0x3fc8fe29  A4      : 0x3fc8b874  A5      : 0x3fc8b87c
	A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fc8fde0  S3      : 0x3fc8ff00
	S4      : 0x00000001  S5      : 0x00000001  S6      : 0x00000000  S7      : 0x00000000
	S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
	T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
	MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
	0x40380001 - _iram_start
	    at ??:??
	MHARTID : 0x00000000
	Stack memory:
	3fc8fdc0: 0x00000000 0x00000000 0x3fc8fdfc 0x4038879a 0x00000000 0x42030b4c 0x3c050250 0x40380030
	0x4038879a - __assert_func
	    at C:\Users\petey\work-temp\idf-test\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-9e04e690dded12b8\out\build\C:/Users/petey/work-temp/idf-test/.embuild/espressif/esp-idf/release-v4.4/components/newlib\assert.c:41
	0x42030b4c - <&T as core::fmt::Display>::fmt
	    at C:\Users\petey\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\fmt\mod.rs:2373
	0x40380030 - _iram_start
	    at ??:??
	3fc8fde0: 0x32303234 0x64313730 0x3fc91000 0x3fc8aba8 0x3fc8fde0 0x3fc8abc4 0x3fc8fddc 0x726f6261
	3fc8fe00: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x30323032 0x20643137
	0x43502074 - _text_end
	
	etc...
	```

Is all that stuff after `abort() was called at...` supposed to still be encoded?

MabezDev, 2022-11-11:
Yes, it doesn't decode stack variables or anything like that. It only decodes the executable code address, which is usually `0x4xxxxx`.

igrr, 2022-11-11:
FYI, it's also possible to decode the actual stack backtrace, instead of feeding the instruction pointers which happen to be on the stack to addr2line. There is a Python based tool in IDF which collects the stack dump and uses GDB to decode the backtrace: https://github.com/espressif/esp-idf/blob/master/tools/gdb_panic_server.py. I think something similar can be easily done in Rust, as well.

mrchantey, 2022-11-11:
Ok that makes sense, the stack overflow i was tracking down was triggering whenever I did a `thread::sleep` so it was hard to follow what was causing the problem.

Anyway it was fixed by a `cargo clean`, thank you for the explanations!

closed 2022-11-11 by mrchantey