esp-rs/espflash #297: Espflash v2 cannot be compiled because of requirement `funty`

lohrm-stabl, 2022-11-16:
Espflash v2 cannot be compiled because of requirement `funty`

```
root@d82fc04d79a3:/workspaces/Rust-SBC-Client# cargo install cargo-espflash --force --version 2.0.0-rc.1
  Downloaded cargo-espflash v2.0.0-rc.1
  Downloaded 1 crate (21.8 KB) in 0.47s
    Updating crates.io index
  Installing cargo-espflash v2.0.0-rc.1
error: failed to compile `cargo-espflash v2.0.0-rc.1`, intermediate artifacts can be found at `/tmp/cargo-installPhnzLR`

Caused by:
  failed to select a version for the requirement `funty = "~1.2"`
  candidate versions found which didn't match: 2.0.0, 1.1.0, 1.0.1, ...
  location searched: crates.io index
  required by package `bitvec v0.22.1`
      ... which satisfies dependency `bitvec = "^0.22.1"` of package `deku v0.14.1`
      ... which satisfies dependency `deku = "^0.14.1"` of package `esp-idf-part v0.1.1`
      ... which satisfies dependency `esp-idf-part = "^0.1.1"` of package `cargo-espflash v2.0.0-rc.1`
```

jessebraham, 2022-11-16:
This package got yanked any they never released a compatible version, see [here](https://github.com/bitvecto-rs/funty/issues/7). Will fix as soon as I'm able.

jessebraham, 2022-12-07:
I've published new versions of `cargo-espflash` and `espflash` (`2.0.0-rc.2`) which should resolve this issue.

closed 2022-12-07 by jessebraham