esp-rs/espflash #298: OTA Example/Documentation

lohrm-stabl, 2022-11-16:
It would be great to have some instructions/examples on how to implement OTA.

Probably the first step would be to build a partition table supporting two OTA partitions, like this:
```csv
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
# bootloader                       32K
# partition table                  3K (4K)
phy_init, data, phy,     0x009000, 4K,
nvs,      data, nvs,     0x00A000, 24K,
ota0,     app, ota_0,    0x010000, 1984K
ota1,     app, ota_1,    0x200000, 1984K,
otadata,  data, ota,     0x3F2000, 8K,
storage,  data, spiffs,  0x3F4000, 48K,
```

Then flash the esp using:
```bash
espflash --partition-table partitions.csv
```

At this point, the initial software version would be flashed.

Then to installing the first update:

I know that I can export the firmware to a `.bin` file using:
```bash
    espflash save-image ESP32 target/{{target}}/debug/sbc-client {{file}}
```
But this would include the bootloader, right? But I don't want to reflash the bootloader. I only want to have the OTA partition. So how would I save an image with only one partition?

The next step is probably to call `espflash write-bin-to-flash`? But using which arguments and how would one specify to only write to the OTA partition?

taunusflieger, 2022-11-16:
You might have a look at my project https://github.com/taunusflieger/anemometer. I have an ota implementation based on esp-idf and scripts which create the application image without the boot loader.

lohrm-stabl, 2022-11-18:
@taunusflieger thanks! That's actually very helpful.
But you use `cargo espflash save-image --release ESP32-S3 --flash-size 2MB ./bin/firmware-$version.bin` (https://github.com/taunusflieger/anemometer/blob/master/create-ota-img.sh) to create an OTA image. But if I read the code right, this would include the bootloader.

taunusflieger, 2022-11-18:
You might be right, but if you are I would not understand why you would create the image without the bootloader. This image gets written into one of the two OTA partitions. The boot process picks up one of the two partitions and loads the image from there. That works with the solution. So either it does not contain the boot loader and it is not required or it contains the boot loader as it is required. Why then would you want to create the image without a boot loader?

dragazo, 2023-05-16:
Any update on this? It looks like there's still not a way to get a binary I can use for OTA updates.

dragazo, 2023-09-19:
From looking at other issues/comments, it seems like the tool always puts the binary in the first app partition? Is that something I can actually rely on? It doesn't seem to be documented anywhere. If so, I think I'll just have to produce a full sized binary, manually parse the partition table, manually slice it out, and hope for the best...

DaneSlattery, 2024-10-25:
https://github.com/DaneSlattery/esp-ota for anyone curious, you could yoink the documentation and examples from here

dragazo, 2024-10-25:
Also updating that I ended up doing the manual approach with a little python script: https://github.com/dragazo/netsblox-vm-esp32/blob/master/extract-ota.py

I still think this should just be a feature of this tool, though. But I'll admit I haven't looked recently - I was just reminded of this when I got a comment ping and thought I'd at least post my solution from way back.

jessebraham, 2025-03-31:
I guess this will be documented in the book, so going to go ahead and close this issue in favour of esp-rs/book#178

closed 2025-03-31 by jessebraham