esp-rs/espflash #316: add defmt support

ImUrX, 2022-12-10:
Adding a flag that parses defmt data would be nice for using with this tool, I currently just pipe the monitor to ``defmt-print``.
I think defmt could also be detected instead of adding a flag but idk which is preferred.

closed 2023-10-11 by SergioGasquez

jjdd12, 2023-11-11:
How is this supposed to work, I am new to using defmt, so I likely be understanding how exactly this works. When using defmt to log, do I have to add any parameters to the monitor command?

This is the output I get when I am expecting to some actual debug logs 
```

~~~~~   ~

```

SergioGasquez, 2023-11-13:
See https://github.com/esp-rs/esp-println?tab=readme-ov-file#defmt for instructions, here is also a simple repo with defmt support: https://github.com/bugadani/esp32s3-defmt-example. We are working on updating the docs so we explain this process better (https://github.com/esp-rs/esp-println/pull/68) and we also add some documentation in espflash

PR #466 (2023-10-11): Print esp-println generated defmt messages
Closes #316 

Counterpart of https://github.com/esp-rs/esp-println/pull/48 as it relies on the injected framing bytes. Not sure how to best lift this curse yet.

This PR adds a new compile-time `defmt` flag that enables defmt support in the binaries and the library. For this, the PR raises MSRV to 1.70. By default, the `defmt` feature is enabled. defmt detection is done by parsing the elf file.

The PR refactors serial port input handling so library users can define their own processing if they so choose. The default processor is able to recognize and [... cut]